lancethepants/tomatoware #9: static compile redsocks2 with tomatoware-arm

rampageX, 2015-04-17:
redsocks2: https://github.com/semigodking/redsocks.git

Compile shared version is fine, but can not run on my EA6900 with tomato and optware installed or entware installed.

So i decide to compile static version, after ask the redsocks2 developer, i use:

CFLAGS=-static
make

But i got these error:

/opt/lib//libcrypto.a(c_zlib.o): In function `zlib_stateful_expand_block':
c_zlib.c:(.text+0x54): undefined reference to `inflate'
/opt/lib//libcrypto.a(c_zlib.o): In function `zlib_stateful_compress_block':
c_zlib.c:(.text+0xd8): undefined reference to `deflate'
/opt/lib//libcrypto.a(c_zlib.o): In function `bio_zlib_free':
c_zlib.c:(.text+0x128): undefined reference to `inflateEnd'
c_zlib.c:(.text+0x144): undefined reference to `deflateEnd'
/opt/lib//libcrypto.a(c_zlib.o): In function `zlib_stateful_finish':
c_zlib.c:(.text+0x194): undefined reference to `inflateEnd'
c_zlib.c:(.text+0x19c): undefined reference to `deflateEnd'
/opt/lib//libcrypto.a(c_zlib.o): In function `zlib_stateful_init':
c_zlib.c:(.text+0x224): undefined reference to `inflateInit_'
c_zlib.c:(.text+0x26c): undefined reference to `deflateInit_'
/opt/lib//libcrypto.a(c_zlib.o): In function `bio_zlib_ctrl':
c_zlib.c:(.text+0x4a0): undefined reference to `deflate'
c_zlib.c:(.text+0x56c): undefined reference to `zError'
/opt/lib//libcrypto.a(c_zlib.o): In function `bio_zlib_write':
c_zlib.c:(.text+0x708): undefined reference to `deflate'
c_zlib.c:(.text+0x794): undefined reference to `zError'
c_zlib.c:(.text+0x824): undefined reference to `deflateInit_'
/opt/lib//libcrypto.a(c_zlib.o): In function `bio_zlib_read':
c_zlib.c:(.text+0x900): undefined reference to `inflate'
c_zlib.c:(.text+0x954): undefined reference to `zError'
c_zlib.c:(.text+0x9e0): undefined reference to `inflateInit_'
collect2: error: ld returned 1 exit status
Makefile:94: recipe for target 'redsocks2' failed
make: *** [redsocks2] Error 1

Any suggest? Thx!

lancethepants, 2015-04-17:
Short Answer: Try running 
"CFLAGS=-static LDFLAGS=-lz make"

Long Answer. Often times when trying to compile applications statically, dependencies that themselves have dependencies sometimes have issues.  In this case, openssl in tomatoware has been compiled to use libz for compression. 

The error above is complaining that some of the functionality that openssl needs is not found.  Googling some of the undefined terms usually will hint as to what library link is missing.  There may be more than one, or many additional libraries that need to be explicitly linked.
The process of linking and resolving dependencies when done dynamically seems to be much more automatic.  Most applications don't consider that you might want to compile statically, so they're Makefiles may not take it into account.

The "undefined reference" error is my favorite error message to encounter. It will usually mean your almost done compiling, and it is usually easy to figure out the missing library links that are needed.

rampageX, 2015-04-17:
More info:
Shared version run on tomatoware: OK

Shared version run on optware:

[root@unknown root]$ /opt/sbin/redsocks2 -v
/opt/sbin/redsocks2: can't resolve symbol '_dl_free_function'

 Shared version run on entware: 

root@unknown:/tmp/home/root# which redsocks2
/opt/sbin/redsocks2
root@unknown:/tmp/home/root# /opt/sbin/redsocks2 -v
-sh: /opt/sbin/redsocks2: not found
root@unknown:/tmp/home/root#

rampageX, 2015-04-17:
@lancethepants thanks! I add -lz to the end of command in Makefile, compile successed!
override LIBS += -lssl -lcrypto -ldl -lz

closed 2015-04-17 by rampageX

lancethepants, 2015-04-17:
Awesome! Glad you figured out a solution.

Ordering can matter when it comes to linking libraries. -lz does need to come after the openssl libs, and now I see LDFLAGS would have placed it before. Unfortunately there is no silver bullet when it comes to static linking, so it's usually having to figure it out case by case (and sometimes hand editing Makefiles).

But now you should have a nicely potable binary.