lancethepants/tomatoware #10: compile redsocks2

houzi-, 2015-11-11:
I compiled redsocks2 with the following warnings, resulting in the compilation interrupt!

````
cc -fPIC -O3 -std=gnu99 -Wall -DUSE_CRYPTO_OPENSSL   -c -o gen/version.o gen/version.c
cc -fPIC -O3 -std=gnu99 -Wall -DUSE_CRYPTO_OPENSSL -o redsocks2 parser.o main.o redsocks.o log.o direct.o ipcache.o autoproxy.o encrypt.o shadowsocks.o http-connect.o socks4.o socks5.o http-relay.o base.o base64.o md5.o http-auth.o utils.o redudp.o socks5-udp.o shadowsocks-udp.o tcpdns.o gen/version.o  -levent -lssl -lcrypto
/opt/lib/gcc/arm-buildroot-linux-uclibcgnueabi/4.9.2/../../../libcrypto.so: warning: gethostbyname is 
obsolescent, use getnameinfo() instead.
```

lancethepants, 2015-11-11:
mipsel or arm?
tomatoware version?
redsocks2 source?

https://github.com/semigodking/redsocks
This compiles just fine for me on arm using the tomatoware 1.0.3.  From what you've posted, there is only compilation warning, but no compilation error.  I get same warning, but the binary is still created just fine. Are you sure resocks2 binary is not there?

houzi-, 2015-11-12:
Yes! tomatoware 1.0.3 arm
But does not generate a binary file

lancethepants, 2015-11-12:
What source code are you using? Can you give me the part of your build log that shows and says 'error'?

houzi-, 2015-11-13:
I am sorry! Prior wrong, in fact file has been generated! And I would like to ask mips gcc version will be updated to 4.8 above it?

lancethepants, 2015-11-13:
Mipsel up to now has been using the entware toolchain. Now that I've found out that entware is being discontinued in favor of entware-ng, I think I will also upgrade the mipsel toolchain like they are.  I'm going to wait until the latest version of buildroot is officially released to do the change.

closed 2015-11-13 by lancethepants