lancethepants/tomatoware #11: static compile shadowsocks-libev with tomatoware-mipsel

houzi-, 2015-11-14:
https://github.com/shadowsocks/shadowsocks-libev
In the mipsel compiler will not pass , and in arm without any problems.
Translation parameters:
```
./configure --prefix=/opt/shadowsocks-libev && make
```
Errors that occur:
```
../libev/.libs/libev.a(ev.o): In function `epoll_init':
/mnt/sda1/opt/build/shadowsocks-libev/libev/ev_epoll.c:242: undefined reference to `epoll_create1'
collect2: ld returned 1 exit status
Makefile:553: recipe for target 'ss-local' failed
make[2]: *** [ss-local] Error 1
make[2]: Leaving directory '/tmp/mnt/sda1/opt/build/shadowsocks-libev/src'
Makefile:488: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/tmp/mnt/sda1/opt/build/shadowsocks-libev'
Makefile:310: recipe for target 'all' failed
make: *** [all] Error 2
```

houzi-, 2015-11-17:
@Look forward to your updates

closed 2015-11-17 by houzi-

lancethepants, 2015-11-24:
Learned a bit about this issue while working on Tomatoware for mipsel

epoll_create1() was added to the kernel in version 2.6.27.  
Mipsel based routers are using kernel 2.6.22.19.

That is the reason the program does not compile. I think you should be able to change all instances of epoll_create1 -> epoll_create, and it should hopefully work.

houzi-, 2015-11-27:
@lancethepants
I feel a bit of trouble, I still other new version uclibc it!
 Uclibc version is too old should be the reason?

houzi-, 2015-11-27:
Tonight I'm going to use entware-ng of toochain to compile a look, it's gcc version 4.8.5

lancethepants, 2015-11-28:
You will have the same issue with entware because they are using the same kernel. It's the lack of the epoll_create1 support in the kernel.  I'm close to releasing tomatoware with uclibc-ng 1.0.9 and gcc 5.2,  and as I suspected, it doesn't compile there either.  

There is however only one occurrence of epoll_create1.  I changed it to epoll_create and it compiles.  Not sure if there are other downsides that could happen because of this.  You should bring up an issue with the shadowsocks author to support older kernels without epoll_create1.

houzi-, 2015-12-11:
@lancethepants 
I understand what you said modified parameters into the epoll_create1 epoll_create, actually compiled a success! Midway some warnings compiler works! Thank you for your guidance! Haha !

And this does not know whether to compile the kernel with related plug-ins , because ipset version tomato_arm nomatch not included on this feature! I want to re -compile ipset latest

houzi-, 2015-12-11:
@lancethepants 
I statically compiled when the problem occurs:
```shellscript
/mmc/lib/gcc/mipsel-buildroot-linux-uclibc/5.2.0/libgcc_eh.a(unwind-dw2-fde-dip.o): In function `_Unwind_Find_FDE':
/home/lance/tomatoware/src/gcc/gcc-build/mipsel-buildroot-linux-uclibc/libgcc/../../../gcc-5.2.0/libgcc/unwind-dw2-fde-dip.c:459: undefined reference to `dl_iterate_phdr'
/home/lance/tomatoware/src/gcc/gcc-build/mipsel-buildroot-linux-uclibc/libgcc/../../../gcc-5.2.0/libgcc/unwind-dw2-fde-dip.c:459: undefined reference to `dl_iterate_phdr'
collect2: error: ld returned 1 exit status
```

lancethepants, 2015-12-11:
A lot of applications don't make things easy to statically compile them.  In most use cases they will be used in linux distributions that almost all link binaries dynamically. Because of that sometimes you have to be creative on how to make it compile in a static manner.

shadowsocks in particular uses a tool called libtool to help with the compiling and linking of the application. libtool is used many places.  In the case of libtool, usually the flag "-all-static" signals to libtool to tell the application to link everything statically.  In those cases I will usually run
./configure --prefix=/mmc
LDFLAGS="-all-static" make

Sometimes you also might need to use the LIBS parameter to help it link necessary libraries as well.  Shadowsocks for some reason doesn't accept LDFLAGS from the command line when run 'make'.  So this is the solution I used.

run
./configure --prefix=..... (whatever prefix you want)
then I edit src/Makefile (which is the directory where those binaries are built), and find the LDFLAGS variable, and set it to "-all-static"
then run make

This for me made them static binaries.

lancethepants, 2015-12-11:
Just to make you aware, tomatoware cannot be used for compiling kernel modules. Kernel modules can only be compiled with the exact toolchain that built the kernel.  Tomatoware can only compile userland applications.  
I though tomato arm was using the newer ipset. From what I remember it should be newer than what is in mipsel firmware.  I know the ad-blocking application in the linksysinfo forum had to modify their program to work with arm ipset.

houzi-, 2015-12-12:
@lancethepants Thank you very much for your guidance, has been a perfect solution

houzi-, 2015-12-12:
@lancethepants Also to bother you help me look at this problem, but also to compile the software errors, this time on the Ubuntu itself with 1.21.0 crosstools-ng compiler cross compiler tool:
（Because using a router to compile a bit slow, and I want to try it, because I use the toochain tomato_arm can be successfully compiled! And mipsel may be due to the GCC version of the old reasons, so you have a build compiler tool! As if the problem is in the latest version of shadowsocks-libev! The old version of the compiler will not appear the problem, really is to disturb you, if you have time to help look at the time!）

/Binutils  /GMP  /MPFR  /ISL  /MPC  /GCC

/2.20.1a  /6.0.0a  /3.1.3  /0.12.2  /1.0.3  /5.2.0
Compiling parameters：
```
CC=mipsel-unknown-linux-uclibc-gcc CXX=mipsel-unknown-linux-uclibc-g++ AR=mipsel-unknown-linux-uclibc-ar RANLIB=mipsel-unknown-linux-uclibc-ranlib ./configure --prefix=/home/mipsel/shadowsocks  --with-openssl=/home/mipsel/openssl --with-zlib=/home/mipsel/zlib --host=mipsel-uclibc-linux
```
ERRO:
```
checking for the pthreads library -lpthreads... no
checking whether pthreads work without any flags... no
checking whether pthreads work with -Kthread... no
checking whether pthreads work with -kthread... no
checking for the pthreads library -llthread... no
checking whether pthreads work with -pthread... no
checking whether pthreads work with -pthreads... no
checking whether pthreads work with -mthreads... no
checking for the pthreads library -lpthread... no
checking whether pthreads work with --thread-safe... no
checking whether pthreads work with -mt... no
checking for pthread-config... no
configure: error: Missing Pthread Library
```
Tips for losing this library file, but this is already contained in the build tool!

lancethepants, 2015-12-14:
I'm not very familiar with crosstools-ng.  I tried to use it for building tomatoware, but I didn't like it. I prefer buildroot.
You've managed to compile it with tomatoware anyway, so I'm not sure why you're looking at an alternate method.

houzi-, 2015-12-14:
@lancethepants
Thank you very much to answer! Why am I doing this? I may be a little incomprehensible reason, I just want to dynamically compile it out and use the built-in dependency libraries tomato, if I use while compiling tomatoware have to have deployed tomatoware otherwise no way to run! Because love does not have some tomato router deployment tomatoware

lancethepants, 2015-12-14:
If you want to build dynamic binaries that use the libraries in tomato, then you shouldn't be using tomatoware or crosstools-ng. You should be using the toolchain that is used to make tomato.  
I prefer static binaries because they can be used in tomato, dd-wrt, openwrt, or RMerlin. Unless you're worried about binary size, I would pick static binaries. I like to store static binaries in /jffs, but now /jffs is so large there is little concern about binary size.  
I additionally compress my binaries with upx to make them smaller.

houzi-, 2015-12-15:
I thought so until this is, but the frustration tomato gcc version is too low, and some function is not supported! Really too worry, static compiler now for me is the best choice, you can use upx to compress and strip again so much smaller than the original!