lancethepants/tomatoware #12: On Mount /mmc compilation errors

houzi-, 2015-12-04:
Before the partition mounted to the /opt and entware coexist not what problem. I try to mount the /mmc compiler error (if not build up the compiler environment)

I should be how to operate.？My command is mounted:
```shellscript
mount -o bind /mnt/sda1/mmc /mmc
```

```shellscript
root@unknown:/mmc/build/openvpn-2.3.2# ./configure --prefix=/mmc/openvpn --enable-pam-dlopen
checking for a BSD-compatible install... ./install-sh -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... no
checking build system type... Warning: creating insecure temp directory
./config.guess: unable to guess system type

This script, last modified 2012-02-10, has failed to recognize
the operating system you are using. It is advised that you
download the most up to date version of the config scripts from

  http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD
and
  http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD

If the version you run (./config.guess) is already up to date, please
send the following data and any information you think might be
pertinent to <config-patches@gnu.org> in order to provide the needed
information to handle your system.

config.guess timestamp = 2012-02-10

uname -m = mips
uname -r = 2.6.22.19
uname -s = Linux
uname -v = #6 Mon Oct 12 14:54:56 CEST 2015

/usr/bin/uname -p =
/bin/uname -X     =

hostinfo               =
/bin/universe          =
/usr/bin/arch -k       =
/bin/arch              =
/usr/bin/oslevel       =
/usr/convex/getsysinfo =

UNAME_MACHINE = mips
UNAME_RELEASE = 2.6.22.19
UNAME_SYSTEM  = Linux
UNAME_VERSION = #6 Mon Oct 12 14:54:56 CEST 2015
```

lancethepants, 2015-12-04:
It might be that the /mmc binaries are not in your path.  
You can check with 'echo $PATH'.
I recommend you copy /mmc/etc/profile to /opt/etc/profile, then re-login through ssh.
http://www.linksysinfo.org/index.php?threads/tomatoware.69742/

houzi-, 2015-12-05:
@lancethepants Thank you for your answer, the problem has been solved! I use the $PATH export to specify, and the entware-ng can coexist with this, right?

lancethepants, 2015-12-07:
Awesome, glad you got it working.  Yes, tomatoware and entware-ng can co-exist.  They just need to be in separate prefixes.
entware-ng => /opt
tomatoware => /mmc

closed 2015-12-07 by lancethepants