lancethepants/tomatoware #13: binutils --enable-plugin

demontanaro, 2015-12-09:
If they have the possibility, please rebuild binutils with the --enable-plugins flag.

There are a lot of programs and libs (alsa, bison, flex, mpfr, etc...) fail to compile and install natively in router from tomatoware.

Compile fail with the error: (in case of GCC 5.3)

sorry - this program has been built without plugin support

The problem is binutils (in tomatoware) has NO built with plugin support (--enable-plugins)

OpenWRT has workaround of these ( a patch) to enable plugins in binutils:

http://openwrt-devel.openwrt.narkive.com/d58b3wl8/patch-binutils-enable-plugin-support

Thanks.

lancethepants, 2015-12-09:
I think the issue you're experiencing is with your environment setup.  Do you have the entware-ng native development packages currently installed? I believe that while you're trying to use tomatoware, some of the entware-ng utilities are being used.  This could be a result of your PATH enviroment variable.  
In tomatoware I provide /mmc/etc/profile, which should replace /opt/etc/profile.  The one important component in it is the PATH variable, which prioritizes tomatoware utilities before entware-ng.

This is the tutorial I've provided, along with libraries and utilities already present in tomatoware.
http://www.linksysinfo.org/index.php?threads/tomatoware.69742/

Make sure your PATH variable has this entware binary directories before entware-ng.  I would even to so far to suggest you either
1. Remove the entware-ng native development packages, or
2. Remove entware-ng from PATH while using tomatoware.

bison, flex, and mpfr are already included in tomatoware and are using their latest versions.
I was able to compile alsa using the following patch
https://raw.githubusercontent.com/buildroot/buildroot/master/package/alsa-lib/0001-add-missing-include.patch
and configuring with './configure --prefix=/mmc --disable-python'
Just to test, I also compiled flex and bison natively with no problems.

I similarly had some issues when I was trying out entware-ng native compiling.  I actually removed tomatoware while I was testing it ensure it did not conflict and try to use the other distributions utilities.

demontanaro, 2015-12-09:
I testing that.
Thanks.

P.D. I'll be back to comment the results.

demontanaro, 2015-12-09:
Nope.
Same results

facts:

Last Entware-ng installed (ando mounted) in /opt 
Tomatoware installed (and mounted) in /mmc
And PATH=
/tmp/home/root # echo $PATH
/mmc/sbin:/mmc/bin:/opt/sbin:/opt/bin:/usr/sbin:/usr/bin:/sbin:/bin:/mmc
/tmp/home/root #

The working directory (when i download the sources) is /opt/tmp (to maintain the tomatoware clean as possible)

Then configure as follows:

./configure --prefix=/mmc --host=mipsel-linux
(and i've proved with prefix=/opt)

Some apps configure and compile correct, others not.

I lost my mind thinking what is the problem...

lancethepants, 2015-12-09:
Don't use "--host=mipsel-linux" and see what that does.  It's not necessary because we're not cross-compiling. Might be the issue.

demontanaro, 2015-12-09:
Good point...

lancethepants, 2015-12-10:
Awesome, glad you got things working.  

I usually recommend creating static binaries, because I personally like portable binaries. Easy to move without requiring tomatoware to run.  All the libraries in Tomatoware come in their static and dynamic varieties.  When compiling library dependencies I'll run "./configure --prefix=/mmc --enable-static --disable-shared. 

Cheers!

closed 2015-12-10 by lancethepants