lancethepants/tomatoware #14: Path on the specified dynamic library

houzi-, 2015-12-11:
A plug-in library path specified dynamic compilation is my "/ mmc / lib", but I want it to path "/ opt / lib", so how should I operate in translation?
```shellscript
        libcrypto.so.1.0.0 => /mmc/lib/libcrypto.so.1.0.0 (0x2aac2000)
        libm.so.1 => /mmc/lib/libm.so.1 (0x2ac8b000)
        libz.so.1 => /mmc/lib/libz.so.1 (0x2acb5000)
        libdl.so.1 => /mmc/lib/libdl.so.1 (0x2acda000)
        libpthread.so.1 => /mmc/lib/libpthread.so.1 (0x2acee000)
        libc.so.1 => /mmc/lib/libc.so.1 (0x2ad16000)
        ld-uClibc.so.1 => /mmc/lib/ld-uClibc.so.1 (0x2aaa8000)
```

lancethepants, 2015-12-11:
Dynamic binaries will always use libraries in /mmc/lib.  /mmc/lib and /mmc/include are the libraries and header files that tomatoware is aware of, and they should always be used together.  If you linked to /opt/lib, while using headers that are found in /mmc/include, there could be incompatibilities.  I use an entirely different, more up-to-date toolchain than entware does, so we can't use their libraries.

You can compile with --prefix=/opt just fine though.  This means that when you run "make install", that the binaries, manpages, and configuration files will be installed in the /opt root, if that's what you prefer.  Despite being installed to /opt, the binaries do and have to use /mmc/lib, because that's what was used to compile the binary.

houzi-, 2015-12-12:
@lancethepants I understand! So I'll have to get a new tool chain, which can be installed in the /opt, for example:
```
Arm-soft-opt.tgz
```

closed 2015-12-12 by houzi-