lancethepants/tomatoware #15: Make an error on ubuntu 14.04

houzi-, 2015-12-16:
version:v1.1-RC
```
make[3]:Leaving the directory `/home/houzi/tomatoware/src/cmake/cmake-3.4.0'
[ 99%] Built target cpack
make[3]: Leaving the directory `/home/houzi/tomatoware/src/cmake/cmake-3.4.0'
Linking CXX executable ../bin/ctest
Error running link command: No such file or directory
make[3]: *** [bin/ctest] erro 2
make[3]:Leaving the directory `/home/houzi/tomatoware/src/cmake/cmake-3.4.0'
make[2]: *** [Source/CMakeFiles/ctest.dir/all] erro 2
make[2]:Leaving the directory `/home/houzi/tomatoware/src/cmake/cmake-3.4.0'
make[1]: *** [all] erro 2
make[1]:Leaving the directory `/home/houzi/tomatoware/src/cmake/cmake-3.4.0'
make: *** [tomatoware] erro 2
```

houzi-, 2015-12-16:
Version V1.1 compiler through

lancethepants, 2015-12-16:
You were able to compile v1.1 fine then?

houzi-, 2015-12-16:
Yes! It is a pre-release version! The official version is no error, it can be very smooth translation

lancethepants, 2015-12-17:
Awesome, glad you were able to compile v1.1.

closed 2015-12-17 by lancethepants