lancethepants/tomatoware #16: Compile shadowVPN Error !

houzi-, 2015-12-23:
```
root@XIAOMI:/mmc/build/ShadowVPN# ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autom4te: need GNU m4 1.4 or later: /usr/bin/m4
aclocal: error: echo failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
```
What is the reason ? Lack of relevant libtool? Also suggests GNU M4 version to date! Very strange

lancethepants, 2015-12-23:
You need to copy or symlink /mmc/etc/profile to /opt/etc/profile, and re-login.
Steps 3 & 4 of Installation here.
http://www.linksysinfo.org/index.php?threads/tomatoware.69742/

There is an environment variable in profile that tells it where to find M4.
export M4=/mmc/bin/m4

lancethepants, 2015-12-23:
Additionally, when cloning that repo, you'll want to do a recursive clone so it also brings in libsodium.

There is however a file you'll manually have to edit.  The file /mmc/libexec/git-core/git-submodule has an inline perl function that tries to use /usr/bin/perl. You'll need to change it to /mmc/bin/perl for it to work. I'll probably have this corrected in future versions.

Then clone with

git clone --recursive https://github.com/ravageralpha/ShadowVPN.git

houzi-, 2015-12-24:
Haha ! It has been resolved! Really strange , I have set before compiling environment
```
export PATH=/mmc/arm-buildroot-linux-uclibcgnueabi/bin/:/mmc/bin/:$PATH
```
Such can not find M4? You must set export m4:
```
export M4=/mmc/bin/m4
```

lancethepants, 2015-12-24:
For some reason autotools and other development packages look for m4 in a hard-coded location, instead of looking for m4 in your PATH.  Fortunately the 'M4' environment variable can be set, and seems to help the development tools find it.  Glad your issue has been resolved.

closed 2015-12-24 by lancethepants

houzi-, 2015-12-24:
understood! Thanks to your answer! there is such a tool (tomatoware) on tomato too convenient! Also it has such a tool (tomatoware) on wanted openwrt