lancethepants/tomatoware #17: Static compiler compile diffutils-3.3 errors

houzi-, 2015-12-30:
```
./configure --prefix=/mmc/diff --enable-static
```
ERRO:
```
dcigettext.c:(.text+0x7e4): undefined reference to `libiconv'
dcigettext.c:(.text+0x9d4): undefined reference to `libiconv_open'
dcigettext.c:(.text+0xb0c): undefined reference to `libiconv_open'
/mmc/lib/gcc/arm-buildroot-linux-uclibcgnueabi/5.2.0/../../../libintl.a(relocatable.o): In function `libintl_set_relocation_prefix':
relocatable.c:(.text+0x44): undefined reference to `libiconv_set_relocation_prefix'
collect2: error: ld returned 1 exit status
Makefile:1222: recipe for target 'cmp' failed
make[2]: *** [cmp] Error 1
```
Compile reason:
I would like to statically compiled diffutils-3.3 so I can quickly mounted to the platform above the other , and tomatoware is built dynamically compiled libraries must rely on to run tomatoware

lancethepants, 2015-12-30:
Always try with LDFLAGS first.

LDFLAGS="-Wl,-static -static -static-libgcc -static-libstdc++" ./configure --prefix=/mmc/diff
make

houzi-, 2015-12-30:
This compiler parameters originally tried! Or dynamic compiler version

houzi-, 2015-12-30:
This parameter is compiled here are dynamic version

lancethepants, 2015-12-30:
Sorry, I'm not sure I understand.

I was able to compile diffutils on the router so that it is static using the configure options I previously posted.

houzi-, 2015-12-31:
it can not be! You can use this parameter compiler to compile a static version? I used the latest tomatoware, but should be okay with this!

rampageX, 2015-12-31:
just use @lancethepants way:

$ file ./src/diff3
./src/diff3: ELF 32-bit LSB executable, ARM, version 1 (SYSV), statically linked, not stripped

houzi-, 2015-12-31:
@rampageX @rampageX 
Thank you very much! Some of the problems I have compiled tomatoware, due to the parameters can not be correctly transferred into @lance tools can be . I previously is compiled using the parameters of thought is to compile parameters are incorrect. Ha ha ha

closed 2015-12-31 by houzi-