lancethepants/tomatoware #18: Tomatoware on DD-WRT

rampageX, 2016-01-04:
I want use Buffalo WZR-1750dhp for compile work because the router have 512M RAM, and only DD-WRT support 512M RAM for now.

Any source without use automake/autoconf were compiled fine, others did not.  When i run `make`,  these error msg show immediately:

Makefile:xxx: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/xxxx/xxxx/xxxx'
Makefile:xxx: recipe for target 'all' failed

I go google and got the same situation post on Open-WRT forum:

https://forum.openwrt.org/viewtopic.php?id=60605

Just remove the trailing '\'  from the up line of the 'dot_seen=no; \'  then 'make' work:

`         fi; \
         dot_seen=no; \`

to

`         fi;
         dot_seen=no; \`

But I need to modify all these Makefile~

Do you have any suggestion? @lancethepants

lancethepants, 2016-01-04:
Hmmm... I'm not sure with this issue. Have you only experienced the issue while using dd-wrt?  Does the same version of Tomatoware work fine for what you're trying to compile with a router loaded with tomato?

rampageX, 2016-01-05:
yes，same tomatoware same program source compile fine on tomato firmware even asus merlin firmware, only happen on dd-wrt...

lancethepants, 2016-01-05:
I build tomatoware using the kernel headers that tomato and rmerlin are built against.  dd-wrt uses a different kernel, so maybe there is a reason there.

rampageX, 2016-01-26:
@darkfires from OpenWrt: (https://forum.openwrt.org/viewtopic.php?id=60605)

>I have figured out a way to (workaround) that issue.  The problem actually has NOTHING to do with make or autotools itself.  I believe the issue is truly in libc, if you compile a static bash on a system with normal glibc, and stick that in /bin/sh, it will solve the make error problem.

So i static compile bash on tomatoware and install, then use:

`CONFIG_SHELL=/mmc/bin/bash /mmc/bin/bash ./configure --prefix=/opt`

and `make` with no error!

rampageX, 2016-01-26:
OK, the easy way is: 

Run `export CONFIG_SHELL=/mmc/bin/bash` first before run `./configure`

It seem like DD-WRT's /bin/bash problem, it's default CONFIG_SHELL. No need to static compile bash, tomatoware include one work fine.

lancethepants, 2016-01-26:
Does dd-wrt really include a real version of bash in /bin/bash?
Does the issue still happen even if you have /mmc/sbin:/mmc/bin first in your $PATH?
(are you using /mmc/etc/profile to setup environment variables, maybe symlinked to /opt/etc/profile).

It might sound like autotools by defualt uses /bin/bash unless the CONFIG_SHELL environment variable is set.
In /mmc/etc/profile I export many environment variables. If you add
```
export CONFIG_SHELL=/mmc/bin/bash
```
to /mmc/etc/profile, making sure it is executing when you login, does this work as well? It think it should, and I can add it there in next release.

rampageX, 2016-01-26:
1. yes, dd-wrt really include a real version of bash in /bin/bash:
```
[tomatoware][compiler:/tmp/mnt/sda2/compile]$ /bin/bash --help
GNU bash, version 4.3.42(1)-release-(arm-openwrt-linux-gnu)
Usage:	/bin/bash [GNU long option] [option] ...
	/bin/bash [GNU long option] [option] script-file ...
GNU long options:
	--debug
	--debugger
	--dump-po-strings
	--dump-strings
	--help
	--init-file
	--login
	--noediting
	--noprofile
	--norc
	--posix
	--rcfile
	--restricted
	--verbose
	--version
Shell options:
	-ilrsD or -c command or -O shopt_option		(invocation only)
	-abefhkmnptuvxBCHP or -o option
Type `/bin/bash -c "help set"' for more information about shell options.
Type `/bin/bash -c help' for more information about shell builtin commands.
Use the `bashbug' command to report bugs.
```
2. yes, /mmc/sbin; /mmc/bin first in my $PATH, the issue still happen, and  when i run `./configure`, it is always generate a Makefile which include `SHELL = /bin/bash` unless the CONFIG_SHELL env is set, it will be `SHELL = /mmc/bin/bash`

3. yes, i add `export CONFIG_SHELL=/mmc/bin/bash` to /mmc/etc/profile, then i relogin, it's work well. (I run `./configure` with different parameters for make sure it generate a new Makefile)

lancethepants, 2016-01-26:
Awesome I will add that that environment variable to 'profile' for future release.
Are you szpunk in linksysinfo?

rampageX, 2016-01-26:
yes，i am 😁

lancethepants, 2016-01-26:
Thanks for your input! I've now committed it to github.

c324a8e58103680fcdce2b28e97a0d0bde01c53e
d6a854ee2f0ef4ebb909c3472c01ec220ae09b4e

closed 2016-01-26 by lancethepants