lancethepants/tomatoware #19: About tomatoware compile busybox problem

houzi-, 2016-01-04:
tomaoware had instructions not compile rely kernel software , is not busybox is to come with the kernel ? 
But I see a condensed version of the compiler on entware-ng!
 I do not quite understand !Also I try to compile busybox tomatoware time error !

lancethepants, 2016-01-04:
Of course you can compile stuff that relies on the kernel, like busybox. It is only kernel modules that you cannot compile.
Tomatoware includes the appropriate kernel headers for both mipsel and arm.

arm -> 2.6.36.4
mipsel -> 2.6.22.19 

Because of the c library uclibc, syncfs(2) is currently not supported, but it looks like in future versions it will be. Until support is added, you have to do the following.

make defconfig
make menuconfig

Go to -> Coreutils
disable the option " Enable -d and -f flags (requres syncfs(2) in libc)"

Additionally, you must disable a few other things if you are compiling on mipsel, because mipsel has an older kernel than arm, and some of the functionality won't work on mipsel's kernel.
For mipsel go to -> Miscellaneous Utilities
and disable the options

i2cget
i2cset
i2cdump
i2cdetect
ubiattach
ubidetach
ubimkvol
ubirmvol
ubirsvol
ubiupdatevol

houzi-, 2016-01-05:
OK! I understand, but I checked crontab compiled but not compiled binary file ?

houzi-, 2016-01-05:
thanks!
ok! i understand

closed 2016-01-05 by houzi-