lancethepants/tomatoware #21: About compiled tomatoware-arm compile wget can not find the makefile mistake!

houzi-, 2016-01-05:
ERROR:
```
root@ASUS:/mmc/build/wget-1.17# make
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
GNUmakefile:106: recipe for target 'abort-due-to-no-makefile' failed
make: *** [abort-due-to-no-makefile] Error 1
```

I try to debug:
```
root@XIAOMI:/mmc/build/wget-1.17# make --debug=v
GNU Make 4.1
Built for arm-unknown-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Reading makefiles...
Reading makefile 'GNUmakefile'...
Reading makefile 'cfg.mk' (search path) (don't care) (no ~ expansion)...
Reading makefile 'maint.mk' (search path) (no ~ expansion)...
Updating goal targets....
Considering target file 'abort-due-to-no-makefile'.
 File 'abort-due-to-no-makefile' does not exist.
 Finished prerequisites of target file 'abort-due-to-no-makefile'.
Must remake target 'abort-due-to-no-makefile'.
There seems to be no Makefile in this directory.
You must run ./configure before running 'make'.
GNUmakefile:106: recipe for target 'abort-due-to-no-makefile' failed
make: *** [abort-due-to-no-makefile] Error 1
```

rampageX, 2016-01-05:
Did you see?!

`You must run ./configure before running 'make'.`

houzi-, 2016-01-05:
@rampageX
GNUmakefile not generated makefile is not first make it! You can try

rampageX, 2016-01-05:
Yes, i try:
`LIBS="-ldl -lz -lintl -liconv -lgmp" LDFLAGS="-Wl,-static -static -static-libgcc" ./configure --prefix=/opt --with-ssl=openssl --disable-nls

make -j2

file ./src/wget
./src/wget: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped
`
So?
When you want static compile something, try dynamic(normal) compile first, and use `ldd` to finger out which libs depend on. Then try @lancethepants 's trick command to static compile. And, read `./configure --help`  careful!

houzi-, 2016-01-06:
@rampageX thanks!
Once again, I find compile time error is not found and unable to generate the makefile gnutls ! But why is it so? Do you know?
Then I mounted directly into / opt instead of / mmc of tomatoware, then the error is completely gone! This is very strange !

lancethepants, 2016-01-09:
Make sure you mount the right version to the right mount point. Are you using the latest version with /mmc prefix? I think I will just distribute /mmc and /mnt from now on, and hopefully that will make everyone happy. 
You were then able to compile the program?

houzi-, 2016-01-10:
I use own translation mount / opt for tomatoware, available for download mount / mmc have problems I encountered

lancethepants, 2016-01-11:
I use the /mmc version that I compile and distribute without having any issues. What version of Tomatoware? and what firmware are you using?  I think the issue is in the setup.

houzi-, 2016-01-12:
I was using the TT-132, I feel should be the settings on it! I then observed to see it! thank you very much

I would like to temporarily close the issues it!

houzi-, 2016-01-21:
@lancethepants 

```
LIBS="-ldl -lz -lintl -liconv -lgmp" LDFLAGS="-Wl,-static -static -static-libgcc" ./configure --prefix=/opt --with-ssl=openssl --disable-nls
```
or 
```
LIBS="-ldl -lz -lintl -liconv -lcrypto -lssl  -lrt -lpthread -lpcre -luuid  " LDFLAGS="-Wl,-static -static -static-libgcc" ./configure --prefix=/opt
```

Use ”arm-soft-mmc“ compile wget appear the same mistakes the last time
(Compiled by myself “arm-soft-opt” no problem)

```
checking for library containing psl_builtin... no
configure: WARNING: *** libpsl was not found. Fallback to builtin cookie checking.
checking for ZLIB... no
checking for compress in -lz... yes
checking for GNUTLS... no
configure: error: Package requirements (gnutls) were not met:

No package 'gnutls' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables GNUTLS_CFLAGS
and GNUTLS_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

lancethepants, 2016-01-27:
On a fresh installation of tomatoware I can create a static `wget` binary using this.
```
LDFLAGS="-Wl,-static -static-libgcc -static-libstdc++" \
./configure \
--prefix=/opt \
--with-ssl=openssl

make LIBS="-lpcre -luuid -lssl -lcrypto -lz -ldl"
```
I prefer using openssl.since it has arm/mipsel asm advantages.

closed 2016-01-27 by houzi-