lancethepants/tomatoware #23: Can not compile GNU M4

houzi-, 2016-01-14:
Because you do not want the dynamic version of tomatoware m4 comes into the / opt / bin, I try to compile it and then the whole dynamic static compiler! But the dynamic compiler will not pass (T_T)

lancethepants, 2016-01-14:
If you want to compile m4, you will need the following patch.
https://github.com/lancethepants/tomatoware/blob/master/patches/m4/gnulib_fix_posixspawn.patch

Why do you need a static version of m4?  I'm getting a little lost in translation, but are you saying you're trying to compile all of tomatoware static?

rampageX, 2016-01-14:
I think he want set up a native compile environment under entware so he can dynamic compile something in it, say: all depend dynamic libs place on `/opt/lib`.
We can do this easy on old Optware because it have `optware-devl` package, but  the new entware lost this package.

lancethepants, 2016-01-14:
Interesting if that is right. Entware-ng does have some support (albeit hagard at best) for native compilation, except for mipsel. As I recall though, @houzi- houzi is using a mipsel router.  

I may have stated in another issue that tomatoware will always use the tomatoware libraries. However, you can use several tricks in the tomatoware compiler to force it to compile against entware-ng.  . If you are interested, I can show you a few of these tricks to make binaries with tomatoware that will dynamic link with entware-ng.  You will need the entware-ng mipsel header files though. Entware-ng don't have a download for the headers files for mipsel though, like they do the other platforms.  You might have to compile all of entware-ng first to get those headers files. Without them, you won't be able to do any native entware-ng native compilation.

houzi-, 2016-01-15:
@rampageX 
@lancethepants 
thanks for your help!
I'm not the intention , I just wanted to try to compile a static version! entware-ng seems to provide gcc compiler environment on the router

closed 2016-01-15 by houzi-