lancethepants/tomatoware #24: Can not be statically compiled curl

houzi-, 2016-01-21:
l try:
```
LIBS="-ldl -lz -lcurl" LDFLAGS="-Wl,-static -static -static-libgcc" ./configure --prefix=/mmc/curl
```
then
l try:
LDFLAGS="-all-static"

same error:
```
/mmc/lib/gcc/arm-buildroot-linux-uclibcgnueabi/5.2.0/../../../../arm-buildroot-linux-uclibcgnueabi/bin/ld: attempted sta
tic link of dynamic object `/mmc/lib/libcurl.so'
collect2: error: ld returned 1 exit status
```

I dynamically compiled database connections needed:
```
libcurl.so.4 => /mmc/lib/libcurl.so.4 (0x4014c000)
libz.so.1 => /mmc/lib/libz.so.1 (0x40126000)
libc.so.1 => /mmc/lib/libc.so.1 (0x401ac000)
libssl.so.1.0.0 => /mmc/lib/libssl.so.1.0.0 (0x4021e000)
libcrypto.so.1.0.0 => /mmc/lib/libcrypto.so.1.0.0 (0x40280000)
ld-uClibc.so.1 => /mmc/lib/ld-uClibc.so.1 (0x40080000)
libdl.so.1 => /mmc/lib/libdl.so.1 (0x40403000)
```

rampageX, 2016-01-21:
remove ` -lcurl` from LIBS

lancethepants, 2016-01-21:
```.
./configure --prefix=/mmc/curl
make LDFLAGS="-all-static" LIBS="-ldl"
```

houzi-, 2016-01-24:
@lancethepants
@rampage
thanks

closed 2016-01-24 by houzi-