lancethepants/tomatoware #25: Questions about the new version ?

houzi-, 2016-01-29:
See tomatoware Added support GO compiled script:
Both mipsel and arm support ?
There is, if you want to overwrite the original tomatoware updated version or re- install unzip it ?
thanks

lancethepants, 2016-01-29:
At this time the build script is only supplied with ARM versions.

For the heck of it I tried it on a mipsel router, but it said `unsupported platform` or something to that effect. There is currently experimental mips64 support in go1.6.... but routers don't have 64bit cpus. Even if it became supported in later versions, I'm not sure how it would be possible to bootstrap the build because go1.5 later are written in go....  For arm we have to first build go1.4+ to then use it to build go1.5+.

Your only option for go on mips is to find a mips cross-compiler with gccgo. Tomatoware of course is a platform that focuses on native compilation so it's outside the scope of this project.  It's something I haven't messed with myself either.

When I upgrade tomatoware (or test it as I do many times in development) I will wipe it out, and extract the new one afterwards.  I have a few things I keep backed up, like my asterisk configuration, and a couple other things.  I actually just scripted up a bash script to restore all my settings and configurations since I do it so much in development.

rampageX, 2016-01-29:
I see someone compile success Go program for mips based router, but only worked on openwrt or dd-wrt, not work on tomato because old kernel used.

lancethepants, 2016-01-29:
I'm guessing they must of used gccgo.  Yeah, dd-wrt and openwrt use newer kernels, so there's a likelihood that something compiled for them wouldn't work on tomato since it might incorporate a feature not found in older kernels.  
I'm certain it would be possible to create a cross-toolchain that could create go binaries for tomato mips.  Someone with more motivation than me I'm sure could do it.  I use and ARM router now so I don't have much interest in mips anymore.

houzi-, 2016-01-29:
Yes, I think the kernel is too low, I am afraid a little difficult to support

lancethepants, 2016-01-30:
I'm sure you could get those same applications to work on mipsel tomato.  You just need to start with the correct toolchain that is aware of the kernel tomato is using, and compile it yourself.  
Right now though it's outside the scope of this project since tomatoware focuses on `native` compilation and not `cross` compilation  We'll see if or when go support mips32 is for a native environment, and go from there.

closed 2016-02-02 by lancethepants