lancethepants/tomatoware #26: tomatoware-mipsel compiler unbound Error

houzi-, 2016-03-11:
Hello! You do not know what is a good solution 
Error:
```
Makefile:266: recipe for target 'ub_event.lo' failed
make: *** [ub_event.lo] Error 1
make: *** Waiting for unfinished jobs....
libtool: compile:  gcc -I. -I/mmc/include -I/mmc/include -I/mmc/include -g -O2 -flto -pthread -c services/outside_network.c -o outside_network.o >/dev/null 2>&1
```

houzi-, 2016-03-12:
arm statically compiled also encountered the same error

rampageX, 2016-03-12:
compile all libs with: 

`./configure --prefix=/mmc --enable-static --disable-shared`

then unbound:

`LDFLAGS="-Wl,-static -static -static-libgcc -s" ./configure --prefix=/opt --with-ssl=/mmc -with-libexpat=/mmc --disable-flto`

got:

    file /opt/sbin/unbound
    /opt/sbin/unbound: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, stripped

houzi-, 2016-03-13:
@rampageX  
Thanks! I try

houzi-, 2016-03-13:
@rampageX 
I do not know why the error occurred , OpenSSL version 1.0.2g
error:
```
configure: error: OpenSSL found in /mmc, but version 0.9.7 or higher is required
```

rampageX, 2016-03-13:
It seem like tomatoware 1.2 bug, you need recompile and install OpenSSL by yourself

houzi-, 2016-03-13:
@rampageX 
Thank you! I do not know you compile OpenSSL is the latest version of it?
I was re-statically compiled OpenSSL, but it is also the error! Remove LDFLAGS can be compiled pass

houzi-, 2016-03-13:
But after` LDFLAGS` can not remove static compiler

rampageX, 2016-03-13:
download and unpack `openssl-1.0.2g.tar.gz` then:

    ./config --prefix=/mmc shared
    make && make install

rampageX, 2016-03-13:
when you `make` you may get error:

`/mmc/lib/gcc/arm-buildroot-linux-uclibcgnueabi/5.2.0/../../../../arm-buildroot-linux-uclibcgnueabi/bin/ld: attempted static link of dynamic object `/mmc/lib/libexpat.so'
collect2: error: ld returned 1 exit status`

run:
    mv /mmc/lib/libexpat.la /mmc/lib/libexpat.la.bak
    mv /mmc/lib/libexpat.so /mmc/lib/libexpat.so.bak

continue `make`, it will done.

houzi-, 2016-03-13:
@rampageX 
Thank you 
It has the perfect solution . OpenSSL is the causes !

lancethepants, 2016-03-14:
OpenSSL in Tomatoware is built with zlib as a dependency for compression.  If you look at the config.log file when it says it couldn't find OpenSSL, you will see that it fails because it can't find the zlib symbols.  You'd have to ad '-lz' somewhere to help it find it.  For some reason unbound's autoconf is really badly behaved and just fights you the whole way.

houzi-, 2016-03-15:
@rampageX Thank you! Now I know! Haha！

houzi-, 2016-03-15:
@lancethepants 
But it is prompted OpenSSL version, rather than not find OpenSSL , OpenSSL version is already higher than this version tips

houzi-, 2016-03-18:
@lancethepants 
```
LDFLAGS="-Wl,-static -static -static-li
bgcc -lz -s" ./configure --prefix=/mmc/unbound-static --with-ssl=/mmc -with-libe
xpat=/mmc --enable-static-exe --enable-static --disable-shared --disable-flto &&
 make LIBS="-all-static -lcrypto -ldl"
```

Now tomatoware-mipsel statically compiled when the error occurs , I have joined -lz but still can not automatically find

error:
```
checking for HMAC_CTX_init in -lcrypto... no
checking if -lcrypto needs -lgdi32... no
checking if -lcrypto needs -ldl... no
configure: error: OpenSSL found in /mmc, but version 0.9.7 or higher is required
```

lancethepants, 2016-03-18:
I try to like to find solutions that can be solved with the command line.  In this case we need the linker flags `-lcrytpo -lz` in that order.  Looking at the configure file, I see that this is impossible at the cli, because anything you put at the command line, LDFLAGS or LIBS, will be placed before `-lcrytpo`, so that doesn't work.

What I did to get this working was modify `configure`, replacing all occurences of `-lcrypto` with `lcrypto -lz -ldl`. I also did the same with `-lssl`, but I don't think it's necessary. I used vim's built in find/replace, or sed would be a good way too (or by hand).

You'll hit the libexpat issue as mentioned above, but you can make that work. It does compile unbound, so the libexpat part is something else.

Now whether or not this will segfault I don't know, please let me know.  

Note: These are not issues with tomatoware.  Most projects do not take static compilation into consideration, so it will not always be a straight-forward solution. Most distros use glibc which does not even allow for static compilation.  Sometimes patching has been needed for static compilation to work.  lighttpd is a famous example.  Here's another example of a cgit patch I needed for static linking to work and for the binary to function properly.

https://lists.zx2c4.com/pipermail/cgit/2015-January/002328.html

houzi-, 2016-03-19:
@lancethepants 
OpenSSL solve the problem , but there have been statically compiled `expat` library file issue
```
/mmc/lib/gcc/mipsel-buildroot-linux-uclibc/5.2.0/../../../../mipsel-buildroot-linux-uclibc/bin/ld: attempted static link of dynamic object `/mmc/lib/libexpat.so'
collect2: error: ld returned 1 exit status
Makefile:303: recipe for target 'unbound' failed
make: *** [unbound] Error 1
```

lancethepants, 2016-03-19:
rampageX has a solution for this up above.  This is really a unbound autoconf issue.

houzi-, 2016-03-20:
But this can not be resolved, or an error

houzi-, 2016-03-20:
@lancethepants
 1.5.9 version of this treatment can be compiled pass
1.5.8 seemingly not pass

lancethepants, 2016-06-17:
Well I released Tomatoware v1.4 with gcc-6.1 you can try out. I successfully compile dozens of static binaries that work fine, so I'm still of the opinion that this is a unbound issue at the core. Have you tried asking them about building static binary?

closed 2016-06-18 by houzi-