lancethepants/tomatoware #27: why my build output is x86 instead of mipsel?

txchen, 2016-03-14:
This maybe a dumb question, but I am not sure if I am using tomatoware correctly:

I downloaded the release file: mipsel-soft-mmc.tgz  extract it to /mmc
then: `export PATH=/mmc/bin:$PATH` so that all the toolchains are from /mmc/bin  

After that, I tried to build shadowsocks (I see qemu-mipsel during build). However, the output target files cannot be run on mipsel environment(the router), but they are runnable on the linux x86 machine.

Am I missing something important? Thanks a lot!

txchen, 2016-03-14:
FYI, this is gcc -v:

```
> gcc -v
Using built-in specs.
COLLECT_GCC=/mmc/bin/gcc
COLLECT_LTO_WRAPPER=/mmc/libexec/gcc/mipsel-buildroot-linux-uclibc/5.2.0/lto-wrapper
Target: mipsel-buildroot-linux-uclibc
Configured with: ../gcc-5.2.0/configure --prefix=/mmc --host=mipsel-buildroot-linux-uclibc --target=mipsel-buildroot-linux-uclibc --with-mpc-include=/home/tomato/tomatoware/mmc/include --with-mpc-lib=/home/tomato/tomatoware/mmc/lib --with-mpfr-include=/home/tomato/tomatoware/mmc/include --with-mpfr-lib=/home/tomato/tomatoware/mmc/lib --with-gmp-include=/home/tomato/tomatoware/mmc/include --with-gmp-lib=/home/tomato/tomatoware/mmc/lib --enable-version-specific-runtime-libs --enable-languages=c,c++ --enable-threads=posix --enable-shared --enable-tls --with-gnu-as --with-gnu-ld --disable-nls --disable-werror --disable-libstdcxx-pch --disable-libssp --with-float=soft --disable-libsanitizer --disable-libgomp --with-arch=mips32 --with-mips-plt
Thread model: posix
gcc version 5.2.0 (GCC)
```

I saw the `--host=mipsel-buildroot-linux-uclibc`, so I did not specify that in my ./configure command line.

txchen, 2016-03-14:
Sorry, my mistake. I can confirm the output is mips

```
ss-redir: ELF 32-bit LSB executable, MIPS, MIPS32 version 1, dynamically linked, interpreter /mmc/lib/ld-uClibc.so.1, not stripped
```

When I run this on my router (rt-n66u), I got `-sh: ./ss-redir: not found`

This might be due to dynamically linking, I will try to figure out how to static compile it.

closed 2016-03-14 by txchen

lancethepants, 2016-03-14:
Very interesting, you're running Tomatoware on x86 hardware, but it's working because you have qemu installed? I've never tried that before, but that's pretty cool.  That's pretty awesome actually, I'll have to give it a try.  Tomatoware is meant to be run on the router itself, 

If you install Tomatoware on /mmc in the router, your binary should then find the libraries.  Or of course you could compile it statically which I find most preferable because of portability.