lancethepants/tomatoware #29: Asterisk 11 and uri rfc3966

sergeynr, 2016-04-16:
Hello Lance! Is it possinble to patch Asterisk 11 RFC 3966? Link: https://issues.asterisk.org/jira/browse/ASTERISK-17179

support RFC 3966 (URI TEL) is necessary for me.

sergeynr, 2016-04-17:
Can you help? Asterisk 11 not support RFC 3966

lancethepants, 2016-04-17:
I've been busy this weekend including today, but I'll take a look at it
this upcoming week, probably tomorrow.

sergeynr, 2016-04-17:
Thanks. i'll be waiting answer. Error Asterisk: [2016-04-16 20:53:07] NOTICE[25402][C-000000b0]: chan_sip.c:18517 check_user_full: From address missing 'sip:', using it anyway
Found peer 'mgts' for 'tel' from 192.168.68.97:5060
== Using SIP RTP TOS bits 184
== Using SIP RTP CoS mark 5
Found RTP audio format 8
Found RTP audio format 18
Found RTP audio format 100
Found audio description format telephone-event for ID 100
Capabilities: us - (ulaw|alaw|g729), peer - audio=(alaw|g729)/video=(nothing)/text=(nothing), combined - (alaw|g729)
Non-codec capabilities (dtmf): us - 0x1 (telephone-event|), peer - 0x1 (telephone-event|), combined - 0x1 (telephone-event|)
Peer audio RTP is at port 192.168.68.101:54588
Peer doesn't provide T.140
[2016-04-16 20:53:07] WARNING[25402][C-000000b0]: chan_sip.c:17630 get_destination: Not a SIP header ()?

<--- Reliably Transmitting (NAT) to 192.168.68.97:5060 --->
SIP/2.0 416 Unsupported URI scheme

lancethepants, 2016-04-18:
What version of tomatoware do you use to test?

sergeynr, 2016-04-18:
version 1.2  
It is Asterisk 11.21.0

Asterisk 13 support RFC 3966

lancethepants, 2016-04-18:
I mean what version of tomatoware?  arm/mipsel, mmc/mnt.  Or I can just push to git and you can compile yourself.

sergeynr, 2016-04-18:
Oh, yes!  Architecture:  arm, mmc

sergeynr, 2016-04-18:
Sorry, I can not compile(

sergeynr, 2016-04-18:
I will be very grateful to you.

lancethepants, 2016-04-18:
http://files.lancethepants.com/Tomatoware/test%201.3%2B/

Give that a test.

sergeynr, 2016-04-18:
Already? Compile with patch? Thanks

sergeynr, 2016-04-18:
It`s work! Many thanks!

lancethepants, 2016-04-19:
Sweet!

closed 2016-04-19 by lancethepants

udadoggdude, 2016-12-08:
@lancethepants Hi Lance,

I'm having the same problem on my asterisk system. Too early for me to want to upgrade the whole system and I don't know if I can get the telco to change 'tel:' to 'sip:'. 

Could you share the patch file again? The link above is not working. 

Many many thanks!

Richie

lancethepants, 2016-12-08:
You can grab a recent nightly file which will have the patch applied to it.
http://files.lancethepants.com/Tomatoware/Nightly/

udadoggdude, 2016-12-08:
Will do. Thanks again.

vahidshatery, 2024-11-12:
> http://files.lancethepants.com/Tomatoware/test%201.3%2B/
> 
> Give that a test.

File not found.
can you upload file again or send to v.shatery@gmail.com .