lancethepants/tomatoware #30: GUI for Asterisk 11

sergeynr, 2016-04-21:
Hello. Whether it is install GUI for Asterisk 11?

lancethepants, 2016-04-25:
Sorry, no gui for asterisk.

To just say no and close I think would be rude, so here's an explanation.

Asterisk shouldn't even be in Tomatoware to start with. It doesn't further or help or extend the purposes of Tomatoware. I put it in one release in order to show case the capability of Tomatoware, and somehow I let it stay in. It's only just there for my convenience, and if others happen to find it useful, awesome. I've probably left it in there too long, and to take it out would break the user experience, so I'll leave it. To extend its capabilities would just add more maintenance and work for something that doesn't belong anyway.

closed 2016-04-25 by lancethepants

sergeynr, 2016-04-26:
Thanks. I understand.