lancethepants/tomatoware #32: How to make a package to know where to found toolchains which tomatoware provided.

zw963, 2016-07-25:
Hi, I have a ASUS RT-AC66U route(MIPS), install package with entware-eg, it worked well.

Though, some package is outdated, e.g. shadowsocks-libev, I want to compile it myself.

I read this issue https://github.com/lancethepants/tomatoware/issues/11, still not help enough.

Following scripts is my attempt guide by the official asuswrt-merlin document here.

```sh

# install older package which need for compile asuswrt-merlin for LinuxMint 17.3
sudo apt-get install bison flex g++ g++-4.4 g++-multilib gawk gcc-multilib \
     gconf-editor gitk lib32z1-dev libncurses5 libncurses5-dev \
     libstdc++6-4.4-dev libtool m4 pkg-config zlib1g-dev gperf lib32z1-dev \
     libelf1:i386 libmpc2:i386

brcm=~/Git/asuswrt-merlin/tools/brcm

# make this script know the old tool chains.
export PATH=$brcm/hndtools-mipsel-linux/bin:$brcm/hndtools-mipsel-uclibc/bin:$PATH
export LD_LIBRARY_PATH=$brcm/hndtools-mipsel-uclibc/lib:/usr/local/lib:/usr/lib
export LIBPATH=$LD_LIBRARY_PATH
export CC=mipsel-linux-uclibc-gcc
export CXX=mipsel-linux-uclibc-g++
export AR=mipsel-linux-uclibc-ar
export RANLIB=mipsel-linux-uclibc-ranlib

# and then, configure and make. compile is passed for me, except, link failed
# seem like GNU tool chains too old and not support by this version shadowsocks-libev
# I create issue for that project, And they told me, need update to a new libc version, 
# or, try to use tomatoware, that maybe worked.
```

I could compile shadowsocks-libev successfully in my local box and remote VPS server.
I just no any idea about how to use tomatoware toolchains to compile MIPS arch
binary for my route, and how to make shadowsocks-libev to know use tomatoware?

Is there exist a guide for this?  Thanks.

rampageX, 2016-07-25:
tomatoware is a way which native compile on router, install and **static** compile your software like shadowsocks-libev, and run it on any OS on router.

zw963, 2016-07-26:
Thanks。

I follow your's guild, first try to build the basic environment which needed.

But I got some error, following is my process

1.  Change config.mk,  export DESTARCH ?= mipsel.

2. make

I got following error:

xz -c -5 < "$inputfile" > archive.dir.tar.xz-t && mv archive.dir.tar.xz-t archive.dir.tar.xz
Opening input file: no such file or directory, /home/Opzw963e/DrnOionppge binnoipxnu/gtc  ofimnmpoiunl/e.te mf:a icnlseo:.  dsn/uopc ho s-fumcihlo defe i.oerll e
d iorre cdtiorreyc,t o/rhyo,m e//hzowm9e6/3z/wD9r6o3p/bDorxo/pcboomxm/ocno/m.meomna/c.se.mda/cpso.-dc/osmtpaartt.-eplo
.el
Makefile:1907: recipe for target 'po-compat.elc' failed
make[5]: *** [po-compat.elc] Error 255
make[5]: *** Waiting for unfinished jobs....
Makefile:1907: recipe for target 'start-po.elc' failed
make[5]: *** [start-po.elc] Error 255
Makefile:1907: recipe for target 'po-mode.elc' failed
make[5]: *** [po-mode.elc] Error 255
make[5]: Leaving directory '/home/zw963/Git/tomatoware/src/gettext/gettext-0.19.8.1/gettext-tools/misc'
Makefile:2026: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/zw963/Git/tomatoware/src/gettext/gettext-0.19.8.1/gettext-tools'
Makefile:1892: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/zw963/Git/tomatoware/src/gettext/gettext-0.19.8.1/gettext-tools'
Makefile:414: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/zw963/Git/tomatoware/src/gettext/gettext-0.19.8.1'
Makefile:370: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/zw963/Git/tomatoware/src/gettext/gettext-0.19.8.1'
Makefile:4: recipe for target 'tomatoware' failed
make: *** [tomatoware] Error 2

zw963, 2016-07-26:
My System is Slackware 14.2,  gcc version 5.3.

lancethepants, 2016-07-26:
If you want to use the tomatoware cross-toolchain, just run `make toolchain`.  The toolchain is located in /opt/tomatoware. 

It looks like you were successfully able to build the toolchain because it continued to compile tomatoware.  If you wish to completely compile tomatoware to then run on your router,  I recommend compiling on Debian, or with Docker using the provided Dockerfile to setup your environment.

If you just want the result of compiling tomatoware, the latest official releases, and also nightly builds are available at http://files.lancethepants.com/Tomatoware/

zw963, 2016-07-26:
Hi, because I am not use this package(gettext-tools), so i tried to comment this in scripts.

compile is continues ...

though,  it broken again finally with followings error messages.

```sh
gnatlink ncurses.ali -L../lib -lAdaCurses -L../../lib -lformw -lmenuw -lpanelw -lncursesw -L/home/zw963/Git/tomatoware/mmc/lib -s -Wl,--dynamic-linker=/mmc/lib/ld-uClibc.so.1 -Wl,-rpath,/mmc/lib -Wl,-rpath-link,/home/zw963/Git/tomatoware/mmc/lib -Wl,-rpath,/home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0/lib -L/home/zw963/Git/tomatoware/mmc/lib -s -Wl,--dynamic-linker=/mmc/lib/ld-uClibc.so.1 -Wl,-rpath,/mmc/lib -Wl,-rpath-link,/home/zw963/Git/tomatoware/mmc/lib
gcc -c -I./ -I. -I../src -I./../src -gnatpn -O2 -I. -I- /home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0/Ada95/src/terminal_interface-curses-text_io-modular_io.adb
gcc -c -I./ -I. -I../src -I./../src -gnatpn -O2 -I. -I- /home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0/Ada95/src/terminal_interface-curses-forms-field_types-user.adb
/usr/lib64/gcc/x86_64-slackware-linux/5.3.0/../../../../x86_64-slackware-linux/bin/ld: ../lib/libAdaCurses.a(static_c_threaded_variables.o): Relocations in generic ELF (EM: 8)
/usr/lib64/gcc/x86_64-slackware-linux/5.3.0/../../../../x86_64-slackware-linux/bin/ld: ../lib/libAdaCurses.a(static_c_threaded_variables.o): Relocations in generic ELF (EM: 8)
../lib/libAdaCurses.a: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/bin/gcc
gnatmake: *** link failed.
Makefile:137: recipe for target 'ncurses' failed
make[3]: *** [ncurses] Error 4
gnatbind -aO. -aO../src -aO./../src -x tour.ali
gnatlink tour.ali -L../lib -lAdaCurses -L../../lib -lformw -lmenuw -lpanelw -lncursesw -L/home/zw963/Git/tomatoware/mmc/lib -s -Wl,--dynamic-linker=/mmc/lib/ld-uClibc.so.1 -Wl,-rpath,/mmc/lib -Wl,-rpath-link,/home/zw963/Git/tomatoware/mmc/lib -Wl,-rpath,/home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0/lib -L/home/zw963/Git/tomatoware/mmc/lib -s -Wl,--dynamic-linker=/mmc/lib/ld-uClibc.so.1 -Wl,-rpath,/mmc/lib -Wl,-rpath-link,/home/zw963/Git/tomatoware/mmc/lib
/usr/lib64/gcc/x86_64-slackware-linux/5.3.0/../../../../x86_64-slackware-linux/bin/ld: ../lib/libAdaCurses.a(static_c_varargs_to_ada.o): Relocations in generic ELF (EM: 8)
/usr/lib64/gcc/x86_64-slackware-linux/5.3.0/../../../../x86_64-slackware-linux/bin/ld: ../lib/libAdaCurses.a(static_c_varargs_to_ada.o): Relocations in generic ELF (EM: 8)
../lib/libAdaCurses.a: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/bin/gcc
gnatmake: *** link failed.
Makefile:140: recipe for target 'tour' failed
make[3]: *** [tour] Error 4
make[3]: Leaving directory '/home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0/Ada95/samples'
Makefile:51: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0/Ada95'
Makefile:113: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/zw963/Git/tomatoware/src/ncurses/ncurses-6.0'
Makefile:4: recipe for target 'tomatoware' failed
make: *** [tomatoware] Error 2
```

zw963, 2016-07-26:
Finally, I tried to compile shadowsocks-libev with following scripts

```sh
set -e
set -x

BASE=`pwd`
SRC=$BASE/src
PATCHES=$BASE/patches
RPATH=$PREFIX/lib
DEST=$BASE/opt
# LDFLAGS="-L$DEST/lib -s -Wl,--dynamic-linker=$PREFIX/lib/ld-uClibc.so.1 -Wl,-rpath,$RPATH -Wl,-rpath-link,$DEST/lib"
LDFLAGS="-Wl,-static -static -static-libgcc -s"
CPPFLAGS="-I$DEST/include"
CFLAGS="-mtune=mips32 -mips32 -O3 -ffunction-sections -fdata-sections"
CXXFLAGS=$CFLAGS
MAKE="make -j`nproc`"
WGET="wget --prefer-family=IPv4"
CPLUS_INCLUDE_PATH=$BASE/mmc/include
C_INCLUDE_PATH=$CPLUS_INCLUDE_PATH

# ./configure --prefix=/opt  

# ./configure --prefix=/opt \
#             --with-ssl=/mmc \
#             -with-libexpat=/mmc \
#             --disable-flto \
#             --host=mipsel-linux \
#             --enable-static --disable-shared

cd ~/Git/shadowsocks-libev/

./configure --prefix=/opt \
            --with-openssl=$BASE/mmc \
            --with-zlib-lib=$BASE/mmc/lib \
            --host=mipsel-linux \
            --enable-static \
            --disable-shared
            # --disable-ssp \

make
```

It passed!. I got correct format binary.  I try to run `ss-redir` in my route, get following error.

```sh
admin@RT-AC66U-20F0:/tmp/home/root# ./ss-redir 
./ss-redir: can't load library 'libm.so.1'
```

I had confuseing by the correct process which I need to do.

If i use docker to setup my compile environment, could you please give me some clue about
how to compile the working version shadowsocks-libev ?

lancethepants, 2016-07-26:
Here's a script to statically compile it.

https://github.com/lancethepants/shadowsocks-libev-mipsel-static

zw963, 2016-07-26:
Cool!

Thanks very much,  I wan to say those scripts is work perfect for slackware 14.2.
And generated binary can be use replace `entware-ng` precompiled version.
worked very well for me..

I will read this script carefully after a while, I belive this could teach me a lot

But, strange enough, after a quick review, I does not found any environment variable
to point out where `tomatoware` is locate, And I try to run this bash scripts from my 
terminal directly, It worked, So, Could you please tell me, what the point about this?
Is this does not use tomatoware totally?
Just set some CFLAGS , configure and LIBS (All what I know from this scripts), why 
we need tomatoware ??

Thanks.

lancethepants, 2016-07-26:
You must have some sort of mipsel cross-toolchain in your path.  I use the tomatoware cross-toolchain found in /opt/tomatoware.  Did you modify your path, or did you install some other cross-compiler?

zw963, 2016-07-26:
Yes, I found the `/opt/tomatoware` directory in my laptop ...

Though, I don't know How and When this directory is created previous.
Because I know I never compile success tomatoware completely.

I have to run make again to know what happen, I understood now.
When run make, the packages which have compile passed will be 
install into /opt/tomatoware automatically.

>>  I use the tomatoware cross-toolchain found in /opt/tomatoware.

Could you please told me, where to use this??
I think I understood bash a little, and write a lot scripts, :laughing:
But I could not found any environment variable refer to /opt/tomatoware ... 
I think shadowsocks-libev should not automatically detect tomatoware too, right?

lancethepants, 2016-07-26:
If you're cross-compiling, you don't need to compile all of tomatoware, you only need the toolchain. If you run just `make`` it will build the toolchain, then continue to build tomatoware.  If you only want a toolchain, you run `make toolchain`.

Like I previously said, somewhere in your $PATH you have a cross-compiler.   Whether it's tomatoware's or another one, I don't know, you will have to look at your $PATH.  You can see what `mipsel-linux-gcc -v` outputs to give you a hint, and `which mipsel-linux-gcc` so see where it's located at.

zw963, 2016-07-26:
I see.  Cool !!

Yes,  I write some startup scripts, will automatic add bin directory inside /opt to $PATH.

```sh
 ╭─ 18:06  zw963 ⮀ ~/Project/asuswrt-merlin-transparent-proxy/route/opt/etc/init.d ⮀ ⭠ (5f5af7c) master $% u= ➦ ruby-2.2.3
 ╰─ $ which -a mipsel-linux-gcc 
/opt/tomatoware/mipsel-soft-mmc/usr/bin/mipsel-linux-gcc

 ╭─ 18:06  zw963 ⮀ ~/Project/asuswrt-merlin-transparent-proxy/route/opt/etc/init.d ⮀ ⭠ (5f5af7c) master $% u= ➦ ruby-2.2.3
 ╰─ $ /opt/tomatoware/mipsel-soft-mmc/usr/bin/mipsel-linux-gcc -v
Using built-in specs.
COLLECT_GCC=/opt/tomatoware/mipsel-soft-mmc/usr/bin/mipsel-linux-gcc.br_real
COLLECT_LTO_WRAPPER=/opt/tomatoware/mipsel-soft-mmc/usr/libexec/gcc/mipsel-buildroot-linux-uclibc/6.1.0/lto-wrapper
Target: mipsel-buildroot-linux-uclibc
Configured with: ./configure --prefix=/opt/tomatoware/mipsel-soft-mmc/usr --sysconfdir=/opt/tomatoware/mipsel-soft-mmc/etc --localstatedir=/opt/tomatoware/mipsel-soft-mmc/var --enable-shared --disable-static --disable-gtk-doc --disable-gtk-doc-html --disable-doc --disable-docs --disable-documentation --disable-debug --with-xmlto=no --with-fop=no --disable-dependency-tracking --target=mipsel-buildroot-linux-uclibc --with-sysroot=/opt/tomatoware/mipsel-soft-mmc/usr/mipsel-buildroot-linux-uclibc/sysroot --disable-__cxa_atexit --with-gnu-ld --disable-libssp --disable-multilib --with-gmp=/opt/tomatoware/mipsel-soft-mmc/usr --with-mpfr=/opt/tomatoware/mipsel-soft-mmc/usr --with-pkgversion='Buildroot 2016.05-dirty' --with-bugurl=http://bugs.buildroot.net/ --disable-libsanitizer --enable-tls --disable-libmudflap --enable-threads --with-mpc=/opt/tomatoware/mipsel-soft-mmc/usr --without-isl --without-cloog --with-float=soft --disable-decimal-float --with-arch=mips32 --with-abi=32 --enable-languages=c --disable-shared --without-headers --disable-threads --with-newlib --disable-largefile --disable-nls
Thread model: single
gcc version 6.1.0 (Buildroot 2016.05-dirty)
```

zw963, 2016-07-26:
problem is resolved, thank you a lot.

closed 2016-07-26 by zw963

zw963, 2016-07-26:
I tried `make toolchain` again, it compile and install successfully.

under `my /opt/tomatoware/mipsel-soft-mmc`,  About 164M files.

In fact, I compare files, found almost the same as previous build.

Though, almost all understood, still exit one more question,

```sh
export PREFIX ?= /mmc
#export PREFIX ?= /mnt
```

Those /mmc or /mnt which should I choice ?  (current keep no changed, /mmc)

lancethepants, 2016-07-26:
PREFIX is the prefix where the compiled application would expect to find its library dependencies running on the target system.  /mmc -> /mmc/lib,  /opt -> /opt/lib.

However, when creating completely static binaries, the prefix does not matter because there are no external dependencies, it is all compiled into the binary.

zw963, 2016-07-27:
Thanks, understood~