lancethepants/tomatoware #34: static compiel tcpdump_4.7.4 error

houzi-, 2016-09-24:
Hi Lance

i use this compile
```
LDFLAGS="-Wl,-static -static -static-libgcc" LIBS="-lcrypto -lpcap"
```
error
```
/mmc/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_globallookup':
dso_dlfcn.c:(.text+0x10): undefined reference to `dlopen'
dso_dlfcn.c:(.text+0x20): undefined reference to `dlsym'
dso_dlfcn.c:(.text+0x2c): undefined reference to `dlclose'
/mmc/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_bind_func':
dso_dlfcn.c:(.text+0x33c): undefined reference to `dlsym'
dso_dlfcn.c:(.text+0x3f0): undefined reference to `dlerror'
/mmc/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_bind_var':
dso_dlfcn.c:(.text+0x48c): undefined reference to `dlsym'
dso_dlfcn.c:(.text+0x53c): undefined reference to `dlerror'
/mmc/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_load':
dso_dlfcn.c:(.text+0x5a8): undefined reference to `dlopen'
dso_dlfcn.c:(.text+0x608): undefined reference to `dlclose'
dso_dlfcn.c:(.text+0x63c): undefined reference to `dlerror'
/mmc/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_pathbyaddr':
dso_dlfcn.c:(.text+0x6cc): undefined reference to `dladdr'
dso_dlfcn.c:(.text+0x72c): undefined reference to `dlerror'
/mmc/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_unload':
dso_dlfcn.c:(.text+0x788): undefined reference to `dlclose'
/mmc/lib/libcrypto.a(c_zlib.o): In function `zlib_stateful_expand_block':
c_zlib.c:(.text+0x54): undefined reference to `inflate'
/mmc/lib/libcrypto.a(c_zlib.o): In function `zlib_stateful_compress_block':
c_zlib.c:(.text+0xd8): undefined reference to `deflate'
/mmc/lib/libcrypto.a(c_zlib.o): In function `bio_zlib_free':
c_zlib.c:(.text+0x124): undefined reference to `inflateEnd'
c_zlib.c:(.text+0x140): undefined reference to `deflateEnd'
/mmc/lib/libcrypto.a(c_zlib.o): In function `zlib_stateful_finish':
c_zlib.c:(.text+0x198): undefined reference to `inflateEnd'
c_zlib.c:(.text+0x1a0): undefined reference to `deflateEnd'
/mmc/lib/libcrypto.a(c_zlib.o): In function `zlib_stateful_init':
c_zlib.c:(.text+0x228): undefined reference to `inflateInit_'
c_zlib.c:(.text+0x270): undefined reference to `deflateInit_'
/mmc/lib/libcrypto.a(c_zlib.o): In function `bio_zlib_ctrl':
c_zlib.c:(.text+0x4a8): undefined reference to `deflate'
c_zlib.c:(.text+0x564): undefined reference to `zError'
/mmc/lib/libcrypto.a(c_zlib.o): In function `bio_zlib_write':
c_zlib.c:(.text+0x6f4): undefined reference to `deflate'
c_zlib.c:(.text+0x780): undefined reference to `zError'
c_zlib.c:(.text+0x80c): undefined reference to `deflateInit_'
/mmc/lib/libcrypto.a(c_zlib.o): In function `bio_zlib_read':
c_zlib.c:(.text+0x8f0): undefined reference to `inflate'
c_zlib.c:(.text+0x94c): undefined reference to `zError'
c_zlib.c:(.text+0x9cc): undefined reference to `inflateInit_'
collect2: error: ld returned 1 exit status
make: *** [Makefile:361: tcpdump] Error 1
```

rampageX, 2016-09-24:
LIBS="-ldl -lz" LDFLAGS="-Wl,-static -static -static-libgcc -s"

lancethepants, 2016-09-26:
`undefined reference to` is my favorite error.   These are symbols that the linker is looking for in order to do the final compilation.  If you look at the errors, it's pretty self explanatory.   You can see it can't find dlopen or dlclose.  These are provided by libdl, or -ldl.  You can also see it's looking for zlib_stateful_finish.  Well right there it tells you 'zlib', so that would be -lz.  Sometimes they're not so obvious. I literally will just google it to see what library which symbols come from.

There's is (hopefully) a planned change to make uclibc more like musl, in that there will be a single system library with all the symbols, this being libc.  So hopefully in the future, some of these we won't have to worry about other system libraries like libdl, pthread, librt etc.

houzi-, 2016-09-28:
@rampageX @lancethepants 
Thanks of all

closed 2016-09-28 by houzi-