lancethepants/tomatoware #37: static compile python-3.5.1

houzi-, 2017-04-17:
Hi, Lance
I am static compiling Python with the following problems:(How should i solve it? )

```
LDFLAGS="-Wl,-static -static -static-libgcc -static-libstdc++ -lintl -s" ./configure --
disable-ipv6 --prefix=/mmc/prthon-3.6.1
```

```
Modules/_localemodule.o: In function `PyIntl_bindtextdomain':
/mmc/compile/Python-3.6.1/./Modules/_localemodule.c:539: undefined reference to `libintl_bindtextdomain'
Modules/_localemodule.o: In function `PyIntl_dcgettext':
/mmc/compile/Python-3.6.1/./Modules/_localemodule.c:495: undefined reference to `libintl_dcgettext'
Modules/_localemodule.o: In function `PyIntl_dgettext':
/mmc/compile/Python-3.6.1/./Modules/_localemodule.c:481: undefined reference to `libintl_dgettext'
Modules/_localemodule.o: In function `PyIntl_gettext':
/mmc/compile/Python-3.6.1/./Modules/_localemodule.c:468: undefined reference to `libintl_gettext'
Modules/_localemodule.o: In function `PyIntl_textdomain':
/mmc/compile/Python-3.6.1/./Modules/_localemodule.c:508: undefined reference to `libintl_textdomain'
collect2: error: ld returned 1 exit status
make: *** [Makefile:736: Programs/_freeze_importlib] Error 1
```

lancethepants, 2017-04-17:
Looking at this I got it to compile with the following.
https://wiki.python.org/moin/BuildStatically

./configure LDFLAGS="-static" --disable-shared
make LDFLAGS="-static" LIBS="-lintl -liconv" LINKFORSHARED=" "

houzi-, 2017-04-18:
Why do these modules fail?

```
Failed to build these modules:
_asyncio              _bisect               _blake2            
_codecs_hk            _codecs_iso2022       _codecs_jp         
_codecs_kr            _codecs_tw            _crypt             
_csv                  _ctypes               _ctypes_test       
_curses               _curses_panel         _datetime          
_decimal              _elementtree          _heapq             
_json                 _lsprof               _multiprocessing   
_opcode               _pickle               _posixsubprocess   
_random               _sha3                 _struct            
_testbuffer           _testcapi             _testimportmultiple
_testmultiphase       array                 audioop            
binascii              cmath                 fcntl              
grp                   math                  mmap               
ossaudiodev           parser                pyexpat            
resource              select                spwd               
syslog                termios               unicodedata        
xxlimited   
```

lancethepants, 2017-04-18:
In the link I posted, it states that it does not by default include any modules.  After running configure you have to edit the "Setup.local" file with the modules you would like it to compile.  You can look at the "Setup" for some examples.  This gets pretty involved.  We're not compiling just some application, this is a programming language, and a fickle one to compile at that.  Is it _ctypes specifically you need?
See if this binary fulfills your needs.
http://files.lancethepants.com/Binaries/python/arm/

houzi-, 2017-04-19:
Use your binary test a bit, there is such a problem:

I set the settings for calling the local library:
`export PYTHONHOME='/jffs/python3.6.1'`

```
root@unknown:/tmp/home/root# python 
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x0073b000 (most recent call first):
Aborted
```

closed 2018-03-28 by lancethepants