lancethepants/tomatoware #39: mipsel cross-compile fails for gcc

HunterZ, 2017-11-11:
Building commit 30280bc fails for me in Debian 8 (jessie) with the output below.

```
make[3]: Entering directory '/home/ben/projects/tomatoware/src/gcc/gcc-build/gcc'
mipsel-buildroot-linux-uclibc-g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -L/home/ben/projects/tomatoware/mmc/tomatoware/lib -s -Wl,--dynamic-linker=/mmc/tomatoware/lib/ld-uClibc.so.1 -Wl,-rpath,/mmc/tomatoware/lib -Wl,-rpath-link,/home/ben/projects/tomatoware/mmc/tomatoware/lib -o lto1 \
	lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a  -L/home/ben/projects/tomatoware/mmc/tomatoware/lib -L/home/ben/projects/tomatoware/mmc/tomatoware/lib -L/home/ben/projects/tomatoware/mmc/tomatoware/lib -lmpc -lmpfr -lgmp -rdynamic  -L./../zlib -lz libcommon.a ../libcpp/libcpp.a  /home/ben/projects/tomatoware/mmc/tomatoware/lib/libiconv.so -Wl,-rpath -Wl,/home/ben/projects/tomatoware/mmc/tomatoware/lib ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
mipsel-buildroot-linux-uclibc-gcc -nostdinc -x c /dev/null -S -o /dev/null -fself-test=../../gcc-7.2.0/gcc/testsuite/selftests
mipsel-buildroot-linux-uclibc-gcc.br_real: error: unrecognized command line option ‘-fself-test=../../gcc-7.2.0/gcc/testsuite/selftests’
Makefile:1933: recipe for target 's-selftest' failed
make[3]: *** [s-selftest] Error 1
```

lancethepants, 2017-11-14:
Occasionally the toolchain to compile tomatoware gets updated, so I just want to check that you've compiled the latest toolchain, which will be running gcc-7.2 that is located in /opt/tomaotware/mipsel-soft-mmc.  If not, you will have to 'rm -rf /opt/tomatoware/mipsel-soft-mmc' and run tomatoware again to compile that new toolchain.

lancethepants, 2017-11-14:
I installed Debian 8 fresh in a VM and installed just the packages mentioned in the README and it all compiled without issue.  There must be something funny about your environment, or maybe the toolchain as I mentioned above.

HunterZ, 2017-11-15:
Yes, I think the toolset was the problem. I think this has bitten me before, but it happens just infrequently enough for met to forget.

It would be really handy if the build process could audit the toolset version and fail and/or rebuild it when it becomes out of date.

lancethepants, 2017-11-16:
I've added a rudimentary check in 45d5292af608ce605da41e11320f1bba4edcb20c

closed 2017-11-16 by lancethepants