lancethepants/tomatoware #40: Some experimental edits

someburner, 2018-02-26:
Just a friendly heads up that I tried out the following with success on my ac3200 ARMv7:

- [lbzip2](http://lbzip2.org/download) - installed/enabled by a configure flag in `tar`. On my laptop it is blazing fast. I havent done formal tests on the router but the online man pages for tar say they plan to use it by default anyways so couldnt hurt
- [boost](http://www.boost.org/) - took me a while to get right since their build system is super confusing in terms of what exactly is required for cross compiling. still a bit hacky how I have it. I guess boost looks for `user-config.jam`, but that it must be in the `$PATH` somewhere, otherwise it looks like `$HOME`. I generally try to avoid messing with the path as much as possible so I just echo what its looking for into a file in home.
- [libtins](http://libtins.github.io/) - a handy libray that hooks into libpcap. depends on some boost header files for regex, maybe some others.

All of these were things I installed natively first using your amazing toolchain. After some thought I figured adding the packages to the xcompile process would be much more maintainable for myself in the case of new/changing routers. Boost is fairly often required by other programs/libraries and I think libtins speaks for itself it terms of usefulness on a router. But not sure if those qualify as "core build tools" and only have ARM hardware to test, hence the fork instead of PR, but I can PR if you want. Also tried using the latest builtroot and gcc 7.3 with no immediate issues as far as I can tell. I assume the patch is no longer needed for 7.3.

Also, a **HUGE** thanks for putting this out there. I just wish I had found it earlier- lots of time wasted on entware. distcc was also new on me and blew my mind a bit. Read through the tomatoware thread on linksysinfo and disagree that this is not a replacement for entware.. Entware takes forever to build and breaks so easily, both when cross compiling and when using stuff it has compiled. They should just give up and work on a distributed version of tomatoware.

lancethepants, 2018-02-27:
I've actually thought about adding boost in the past, but was kind of waiting to see if I or anybody else needed it. I'll probably add that, and lbzip stuff, and maybe the libtins.  What is it you compile with those libaries?, just out of curiosity.  Do you create a static binary?  The gcc patch is necessary so that dynamically linked binaries properly point to the tomatoware dynamic linker, instead of the tomato system one.  Doesn't matter for static binaries, but can break things with dynamic ones.

someburner, 2018-02-27:
Okay cool!

For gcc, it looked like 7.3 defaulted to the patch, but I'm not totally sure? There are 2 patches and I left 1. I dont need 7.3 for any reason specifically so I may just go back to whats tried and true.

I was doing static + dynamic for both, or thought I wasm but after looking over it again I think the dynamic tins was broken and Im double checking boost ATM. tins is a great library but not as cross-compile friendly. It does not seem to honor custom install directories. lbzip2 + tar is definely all good for what I have up right now.

A couple other things I was wondering:

* what determines which libs need to be stripped and which dont on the so's?
* been having issues with tar trying to extract the `tbz`. This may be becuase I have some ACL/FS attribute stuff turned on for my kernel (ext4 btw, usb3 SSD). It throws non-fatal warnings about symlink targets not existing. It all extracts and appears to work fine, but just wondering what the deal is on that. Also the `1777` chmod at the end- is that critical?

lancethepants, 2018-02-27:
I'll definitely make the jump to gcc 7.3, but if the patch doesn't apply cleanly I'll have to refactor it a bit. Buildroot should be making a final release in just a few days, so I'll probably do it then.  libtins allows you to tell it the location of libpcap, but not of openssl or boost, so it's picking up my system's openssl instead of tomatoware's.  Might be able to patch, or could say something to the devs to make cross-compilation easier.  cmake was supposed to solve all the woes of cross-compilation, but I think it's pretty terrible and hate seeing anything use it :).  autoconf is much easier and better imo.

I don't explicitly strip any libraries I think. I do have '-s' in my LDFLAGS, but not sure if that gets applied for libraries, or just binaries.  Did you notice the same tbz issue before the new library?  the 1777 is just so that all non-root users have access to it if needed, it's what linux distros do as well.  It probably won't break anything if you have to remove it.  What OS do you use to build tomatoware?

someburner, 2018-02-27:
So I just got it working finally. I ended up retreating back to gcc 7.2 and things are pretty good Id say. If you can look past the edits i made you can probably snag what you need in `base.sh` at the bottom. `lbzip2` at the top and then you need to just change the configure for `tar` to be `=lbzip2`.

I agree that cmake is awful. If it was just a matter of being buggy thats one thing, but it really just introuces huge amounts of dependencies that have to be tweaked to perfection. seems geared towards non-linux users so of course it is going ot be mediocre. I build on ubuntu 17.10 right now. I also use `lxd` to build the base image from my personal rendition of shibby/kille72 fw and will probably just turn this into a container too since they are so convenient.

someburner, 2018-02-27:
### static + shared after fresh mmc extract

```sh
ssh tomato

## get some examples
wget http://libtins.github.io/examples/code/arp-monitor.cpp
wget http://libtins.github.io/examples/code/beacon-display.cpp
wget http://libtins.github.io/examples/code/http-requests.cpp

## watch for arp on eth2
./arp eth2
## 802.11 stuff, not enabled
./beacon eth2
## example of tins + boost_regex, looks for POSTs/GETs
./http br0

## static/dynamic

### Dynamic
g++ -L/mmc/lib -L/mmc/usr/lib -ltins -lpcap arp-monitor.cpp -o arp
g++ -L/mmc/lib -L/mmc/usr/lib -ltins -lpcap beacon-display.cpp -o beacon
g++ -I/mmc/include -L/mmc/lib -L/mmc/usr/lib -ltins -lpcap -lboost_regex http-requests.cpp -o http

root@tomato:~/test/tins-examples# file {arp,beacon,http}
arp:    ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /mmc/lib/ld-uClibc.so.1, not stripped
beacon: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /mmc/lib/ld-uClibc.so.1, not stripped
http:   ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), dynamically linked, interpreter /mmc/lib/ld-uClibc.so.1, not stripped

46K Feb 27 06:15 beacon
282K Feb 27 06:19 http
56K Feb 27 06:13 arp

root@tomato:~/test/tins-examples# ldd ./http
	libtins.so.4.0 => /mmc/lib/libtins.so.4.0 (0x4016e000)
	libpcap.so.1 => /mmc/lib/libpcap.so.1 (0x4008c000)
	libboost_regex.so.1.66.0 => /mmc/lib/libboost_regex.so.1.66.0 (0x4020e000)
	libstdc++.so.6 => /mmc/lib/libstdc++.so.6 (0x402e2000)
	libgcc_s.so.1 => /mmc/lib/libgcc_s.so.1 (0x40124000)
	libc.so.0 => /mmc/lib/libc.so.0 (0x40423000)
	ld-uClibc.so.1 => /mmc/lib/ld-uClibc.so.1 (0x40002000)

### Static
g++ arp-monitor.cpp -o arp -std=c++17 -static -ltins -lpcap
g++ beacon-display.cpp -o beacon -std=c++17 -static -ltins -lpcap
g++ http-requests.cpp -o http -std=c++17 -static -ltins -lpcap -lboost_regex

13M Feb 27 06:32 arp
13M Feb 27 06:36 beacon
14M Feb 27 06:35 http

root@tomato:~/test/tins-examples# file {arp,beacon,http}
file {arp,beacon,http}                                 
arp:    ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped
beacon: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped
http:   ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), statically linked, with debug_info, not stripped
```

someburner, 2018-02-27:
> Did you notice the same tbz issue before the new library? the 1777 is just so that all non-root users have access to it if needed, it's what linux distros do as well. 

Ya its always been like that. but Ive been using some non standard kernel settings for tomato for a while since I needed `netem` loaded, so i think its just a quirk with linked files coming out ahead of where they point. But I also have bash as default shell and turned on a bunch of caching features, so I think I'll just try to ignore the warnings :)

lancethepants, 2018-02-27:
You can strip those static binaries and they will probably come down a lot in size.  I also bundle upx with tomatoware, so you can run something like "upx --ultra-brute ./binary" to further shrink them to ~1/3 of their size.

someburner, 2018-02-27:
nice- thanks for the heads up. I have a 120gb ssd hooked up so space isnt lacking, but lots of swap then too, does static + stipped perform better you think?

lancethepants, 2018-02-27:
It might load faster, but I don't think performance would be any better. So I pushed a commit for boost. On the router I compile libtins, but it gives my "missing symbols" for the dynamically linker library, but works for my on the statically linke one.

someburner, 2018-02-27:
Awesome- thanks for adding it. What are you using for the source for tins? I got those errors as well and it was some combination of the generated `build/cmakelists_install.txt` not getting set to `/mmc` for prefix due to cmake, not having `LD_FLAGS` right (lib installed in /mmc/usr/local or something), or not having just the right order for `-ltins -lpcap`, or not having `-std=c++11` or similar. Also at one point cmake was failing on CXX tests, erroneously disabling the C++ flag because it couldnt detect `chronos` and another C11 feature. So I had to grease the wheels. But given all of that unreliabilty probably not worthy of inclusion. Most people would be fine with/prefer tcpdump anyway.

closed 2018-03-28 by lancethepants