lancethepants/tomatoware #42: how to static compile kdig

rampageX, 2019-01-21:
@lancethepants I tried static compile kdig 2days but failed..so I need you help ~

My process:

> #libedit
> ./configure --prefix=/mmc/s --enable-shared=no
> 
> #libunistring
> ./configure --prefix=/mmc/s --enable-shared=no
> 
> #gmp
> ./configure --prefix=/mmc/s --enable-shared=no
> 
> ##nettle
> ./configure --prefix=/mmc/s --disable-openssl --disable-shared --disable-documentation --with-lib-path="/mmc/lib"
> 
> #gnutls: failed
> LDFLAGS="-lintl -lnettle -lhogweed -lgmp -ldl -Wl,-static -static -static-libgcc -s" NETTLE_CFLAGS="-I/mmc/s/include" NETTLE_LIBS="-L/mmc/s/lib" HOGWEED_CFLAGS="-I/mmc/s/include" HOGWEED_LIBS="-L/mmc/lib" GMP_CFLAGS="-I/mmc/s/include" GMP_LIBS="-L/mmc/s/lib" ./configure --prefix=/mmc/s --enable-static=yes --enable-shared=no --with-default-trust-store-file=/mmc/ssl/certs/ca-certificates.crt --with-included-libtasn1 --without-p11-kit --disable-openssl-compatibility --disable-heartbeat-support --disable-libdane --disable-doc --disable-srp-authentication --disable-psk-authentication --disable-full-test-suite --without-idn
> 
> #knot-dns: failed
> LIBS="-lintl -lnettle -lgmp -lunistring -ldl -lz" LDFLAGS="-Wl,-static -static -static-libgcc -s" ./configure --prefix=/opt --disable-daemon --disable-modules --disable-documentation --disable-fastparser --without-libidn

Any suggest?

lancethepants, 2019-01-21:
This one is definitely tricky. This also shows why I hate libtool, because it strips a lot of the necessary stuff you need to pass to get this to compile.

So all I did was compile libedit and libunistring, because tomatoware already has gmp, nettle, and gnutls.  Maybe you recompiled these for different or additional options.

Then I configured with knot-dns, using part of your configure.
```
./configure \
--prefix=/opt \
--disable-daemon \
--disable-modules \
--disable-documentation \
--disable-fastparser \
--without-libidn
```
Then start compilation with 
```
make V=99 \
CC="distcc arm-linux-gcc" \
CXX="distcc arm-linux-g++" \
LDFLAGS="-zmuldefs -all-static"
```
V=99 so we can see all what stupid libtool is doing. -zmuldefs because we get mutliple definition errors. Fortunately kdig is the first things to fail, at least for me.  It will fail on command similar to
```
libtool: link: distcc arm-linux-gcc -g -O2 -Wall -Werror=format-security -Werror=implicit -Wstrict-prototypes -zmuldefs -static -o kdig utils/kdig/kdig-kdig_exec.o utils/kdig/kdig-kdig_main.o utils/kdig/kdig-kdig_params.o  ./.libs/libknotus.a /mmc/src/knot/knot-2.7.5/src/.libs/libknot.a -L/mmc/lib /mmc/src/knot/knot-2.7.5/src/.libs/libdnssec.a /mmc/lib/libedit.a -lncurses /mmc/lib/libgnutls.a -lnettle -lhogweed /mmc/lib/libgmp.a /mmc/lib/libintl.a /mmc/lib/libiconv.a -lc
```
Copy this command. Delete everything before gcc. We will manually run the command. We'll also surround external libraries with `-Wl,--whole-archive` & `-Wl,--no-whole-archive`.  For some reason sometimes on large projects when static linking many libraries, I have to surround the libraries with this. Maybe some symbols are getting left out, and then the resulting binary will segfault.
cd into ./src
Then the command I run is
```
gcc  -g -O2 -Wall -Werror=format-security -Werror=implicit -Wstrict-prototypes -zmuldefs -static -o kdig utils/kdig/kdig-kdig_exec.o utils/kdig/kdig-kdig_main.o utils/kdig/kdig-kdig_params.o  ./.libs/libknotus.a /mmc/src/knot/knot-2.7.5/src/.libs/libknot.a -L/mmc/lib /mmc/src/knot/knot-2.7.5/src/.libs/libdnssec.a -Wl,--whole-archive /mmc/lib/libedit.a -lncurses /mmc/lib/libgnutls.a -lnettle -lhogweed /mmc/lib/libgmp.a /mmc/lib/libintl.a /mmc/lib/libiconv.a -lc -Wl,--no-whole-archive
```

rampageX, 2019-01-22:
@lancethepants Thanks a lot for your help, i can build the static version of `kdig`  finally. When i test with:
`kdig -d @1.1.1.1 +tls-ca +tls-host=cloudflare-dns.com twitter.com`
I got error:

> ;; DEBUG: Querying for owner(twitter.com.), class(1), type(1), server(1.1.1.1), port(853), protocol(TCP)
> ;; WARNING: TLS, failed to import system certificates (GNUTLS_E_UNIMPLEMENTED_FEATURE)
> ;; ERROR: failed to query server 1.1.1.1@853(TCP)

So i remember how i had to recompile GNUTLS at the first time. Tomatoware include gnutls compile with `--without-p11-kit`, so no ca-file support. I recompile with:

```
$CONFIGURE \
	--enable-local-libopts \
	--without-p11-kit \
	--with-included-libtasn1 \
	--enable-static \
	--disable-doc \
	--with-included-unistring
        --with-default-trust-store-file=/mmc/ssl/certs/ca-certificates.crt 
        --with-default-trust-store-dir=/mmc/ssl/certs
```
then static compile `kdig`, it's working now:

> ./kdig -d @1.1.1.1 +tls-ca +tls-host=cloudflare-dns.com twitter.com
> ;; DEBUG: Querying for owner(twitter.com.), class(1), type(1), server(1.1.1.1), port(853), protocol(TCP)
> ;; DEBUG: TLS, imported 133 system certificates
> ;; DEBUG: TLS, received certificate hierarchy:
> ;; DEBUG:  #1, C=US,ST=California,L=San Francisco,O=Cloudflare\, Inc.,CN=cloudflare-dns.com
> ;; DEBUG:      SHA-256 PIN: TdBczz+YjD3Q/taSfHXL5n4LnRxzJk0WG0JAX7nRu6s=
> ;; DEBUG:  #2, C=US,O=DigiCert Inc,CN=DigiCert ECC Secure Server CA
> ;; DEBUG:      SHA-256 PIN: PZXN3lRAy+8tBKk2Ox6F7jIlnzr2Yzmwqc3JnyfXoCw=
> ;; DEBUG: TLS, skipping certificate PIN check
> ;; DEBUG: TLS, The certificate is trusted.
> ;; TLS session (TLS1.3)-(ECDHE-SECP256R1)-(ECDSA-SECP256R1-SHA256)-(AES-256-GCM)
> ;; ->>HEADER<<- opcode: QUERY; status: NOERROR; id: 22111
> ;; Flags: qr rd ra; QUERY: 1; ANSWER: 2; AUTHORITY: 0; ADDITIONAL: 1
> 
> ;; EDNS PSEUDOSECTION:
> ;; Version: 0; flags: ; UDP size: 1452 B; ext-rcode: NOERROR
> ;; PADDING: 392 B
> 
> ;; QUESTION SECTION:
> ;; twitter.com.        		IN	A
> 
> ;; ANSWER SECTION:
> twitter.com.        	728	IN	A	104.244.42.1
> twitter.com.        	728	IN	A	104.244.42.129
> 
> ;; Received 468 B
> ;; Time 2019-01-22 14:54:38 CST
> ;; From 1.1.1.1@853(TCP) in 53.1 ms

Yes, this is not very portable and real static because the ca-files. These files not exist in normal tomato firmware but new `freshtomato`，it's stay at /rom/cacert.pem then support new DNS-Over-TLS client `Stubby`. So maybe we can specified `--with-default-trust-store-file=` to this file, or Entware's ca-file: /opt/etc/ssl/certs/ca-certificates.crt.

lancethepants, 2019-01-22:
https://www.knot-dns.cz/docs/2.6/html/man_kdig.html

It looks like kdig has a runtime option to provide your own certificate.
+[no]tls-ca[=FILE]

See if that works for you. Then you don't have to worry about re-compiling any of the crypto libraries, and just use tls-ca= option in kdig.

lancethepants, 2019-01-22:
`kdig -d @1.1.1.1 +tls-ca +tls-host=cloudflare-dns.com twitter.com`
Looks like you already provide `+tls-ca` in command, just add =/path/to/cert

rampageX, 2019-01-23:
Ye, thanks!

closed 2019-01-23 by rampageX