lancethepants/tomatoware #43: commit on Sept 4 causes mipsel-cross-compilation on ubound

leandroong, 2019-09-05:
tested on unbound-1.9.1, unbound-1.9.2, unbound-1.9.3.
Error seems to fail to detect correct SSL version. Kindly investigate. TY

lancethepants, 2019-09-05:
./configure --help

--with-ssl=/mmc
--with-libexpat=/mmc

Then it will find those libraries

leandroong, 2019-09-07:
hello again, posting finding update. Using my old tomatoware (v1.7) has compiled successfully same unbound  without issue. 
Regarding tomatoware (v.18), lowering down GCC to 9.1.0 produce same error. I can attest up to certain commits, that unbound is compiling also. 
Maybe buildroot has the issue:
v1.7 - buildroot-2018.08.tar.bz2
v1.8- buildroot-2019.02.tar.bz2
Anyway, feel free to close this issue, thanks again

leandroong, 2019-09-08:
Updates again. I was able to roll back and compile to tag v1.8. Unbound compile successfully. Kindly close. Thanks again

closed 2019-09-09 by lancethepants