lancethepants/tomatoware #51: v3.0 cmake error

rampageX, 2021-09-21:
```
[T][P][netgear:/mmc]$ cmake

/mmc/bin/cmake: symbol '_ZSt28__throw_bad_array_new_lengthv': can't resolve symbol

/mmc/bin/cmake: symbol '_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEv': can't resolve symbol
```

and newest nightly building have this error too.

had google some place, maybe GCC 11 problem?

rampageX, 2021-09-21:
rollback to nightly 07-09-2021, GCC 10.3 fix this issue, all other version using GCC 11 not work.

lancethepants, 2021-09-21:
hmmm, I'm not seeing this on arm or mipsel.  What router and firmware version are you using? Do you wipe out the previous installation before extracting the new one? or do you extract over top the existing installation?

rampageX, 2021-09-21:
Netgear R7000 with FreshTomato 2021.5 ARM. 

All GCC 11 version i just extract over top the existing installation (like i did before many version upgrade... ); 

The nightly 07-09-2021, i wipe out old version and did a fresh extract, because i think many libs/app will be downgrade.

rampageX, 2021-09-21:
BTW: I can live with this version until your upgrade OpenSSL to 3.0 :)

lancethepants, 2021-09-21:
I'm using the same exact router and firmware version. I always wipe out the previous installation before extracting the new one, so maybe that is the issue you're experiencing. I'll try and look at OpenSSL soon.

closed 2021-09-21 by rampageX

lancethepants, 2021-09-24:
OpenSSL now has been upgraded.