lancethepants/tomatoware #52: UPX static compile error

houzi-, 2022-05-05:
Hi guys

error:
````
step1: 0x09f003f7 0x09ec03f6 0x09f003f7
./upx.out: main.cpp: 1364: void upx_sanity_check(): Assertion `upx_adler32(d, 4) == 0x062e0277' failed.
Aborted
````

compile operation:
UCL_1.0.3: ```` ./configure --prefix=/mmc CPPFLAGS="$CPPFLAGS -std=c90 -fPIC"````
LZMA_14.32.7:  ````./configure --prefix=/mmc````
UPX:  ```` make all LIBS="--static -lucl -lz" CXXFLAGS="-Wno-erro"````
````

The above errors will appear when the binary**(upx.out)** file generated by compilation runs. What is the reason for this?

lancethepants, 2022-05-05:
What version of upx are you compiling?
In Tomatoware I've intentionally held back upx to version 3.94 because 3.95 & 3.96 have been buggy with either arm or mipsel.  I did try compiling the very latest from git a few months ago and it seemed to be in better shape.

lancethepants, 2022-05-05:
So usually I cross-compile upx, but this is how to do it natively on your router. This is the method they explain in README.SRC.  You only need ucl and upx.
```
# UCL
CFLAGS="-std=c90" \
./configure

make -j2

# Adjust to your location of ucl.
export UPX_UCLDIR=/mmc/src/upx/ucl-1.03

# UPX
LDFLAGS="-s -static" \
make -j2 \
all \
CXXFLAGS_WERROR= \
CHECK_WHITESPACE=/bin/true
```

houzi-, 2022-05-06:
It's the same mistake. It's estimated that I modified the source code of upx    thanks guys

closed 2022-05-07 by houzi-