lancethepants/tomatoware #53: questions about this toolchain

houzi-, 2022-05-17:
Hi guys

The binaries compiled by this toolchain should be based on ARMV7, right?

lancethepants, 2022-05-17:
Ye

> Hi guys
> 
> The binaries compiled by this toolchain should be based on ARMV7, right?

Yes, armv7 Cortex-A9 softfloat.

houzi-, 2022-05-17:
OK，Will it support armv8 Cortex-A9 in the future?

lancethepants, 2022-05-17:
I would love to make an armv8/arm64/aarch64 toolchain version but I don't have any arm64 hardware.

lancethepants, 2022-05-17:
What arm64 hardware are you running? Just out of curiosity.

houzi-, 2022-05-17:
ASUS AX86U😬

houzi-, 2022-05-17:
merlin

lancethepants, 2022-05-23:
Apparently AsusWRT Merlin userspace tools are using 32bit with no vfpv3 or neon. There is a pending pull request to change this.
https://github.com/RMerl/asuswrt-merlin.ng/pull/804

houzi-, 2022-05-24:
Yes, but the router system of the Merlin Broadcom solution is based on 64 kernels and is compatible with 32-bit and 64-bit binary operation.

lancethepants, 2022-08-04:
Just bought me an rt-ax88u. So may be a few weeks, but I plan to get tomatoware working on this. Would be interesting to see 64bit userspace performance vs 32bit. Not sure why AsusWRT doesn't use 64bit in userspace. Wonder if it's to simplify userspace across all devices.

lancethepants, 2022-08-18:
I've now got Tomatoware functioning on arm64.  Still fixing bugs as I discover them, but it seems to be pretty functional.  I made the switch to musl libc for arm64 because uclibc-ng isn't fully functioning on this architecture. Available right now in the nightly builds, let me know how it works for you.
https://files.lancethepants.com/Tomatoware/Nightly/08-17-2022/

lancethepants, 2022-08-18:
Also got rust running as well.  Just `apt update` and `apt install rust`

houzi-, 2022-08-19:
nice jod.I will try