lancethepants/tomatoware #54: v5 error "recompile with -fPIC"

rampageX, 2022-09-15:
I updated to v5 version, ARM, Most worked apps before compile failed now with these error:
`relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC`

It can be fixed by add `-fPIC` into Makefile's CFLAGS.

Few app include `CFLAGS += -fPIC` in Makefile seen be well.

lancethepants, 2022-09-15:
Hmmm, are you using my release or is it self-compiled version of Tomatoware? If you're using a self-compiled version, make sure to delete and recompile the toolchain. There have been a log of changes and toolchain.sh only checks if the version has changed where other changes could have happened it wouldn't detect.
Do you have a sample program I could try to compile myself?

rampageX, 2022-09-21:
@lancethepants I use your release, you can try `c-ares-1.18.1`, i got:

```
...
 CCLD     ahost
/mmc/lib/gcc/arm-tomatoware-linux-uclibcgnueabi/12.2.0/../../../../arm-tomatoware-linux-uclibcgnueabi/bin/ld: ahost-ahost.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/mmc/lib/gcc/arm-tomatoware-linux-uclibcgnueabi/12.2.0/../../../../arm-tomatoware-linux-uclibcgnueabi/bin/ld: ahost-ares_getopt.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/mmc/lib/gcc/arm-tomatoware-linux-uclibcgnueabi/12.2.0/../../../../arm-tomatoware-linux-uclibcgnueabi/bin/ld: ahost-ahost.o(.text+0x48): unresolvable R_ARM_CALL relocation against symbol `memset'
```

lancethepants, 2022-09-22:
I haven't been able to duplicate this. I downloaded my v5.0 release for arm and was able to compile c-ares using 
`./configure --prefix=/mmc` without issue.
Are you deleting the previous tomatoware installation or extracting over top of it? I always delete the previous installation before extracting.

rampageX, 2022-09-22:
Yes, i deleted the previous installation before extracting, and i even disable distcc this time, but the error still there...  Back to version 4.0, there is no problem。

lancethepants, 2022-09-23:
What configure options are you using?

lancethepants, 2022-09-23:
Maybe also try the latest nightly version. Things are changing a bit so there now is a uclibc and musl version for arm. musl seems to work pretty good but also less tested.

rampageX, 2022-09-23:
> What configure options are you using?

`./configure --prefix=/mmc` only this.

rampageX, 2022-09-23:
> Maybe also try the latest nightly version. Things are changing a bit so there now is a uclibc and musl version for arm. musl seems to work pretty good but also less tested.

I'll try nightly musl version later, does toolchain for distcc need recompile with musl option?

lancethepants, 2022-09-23:
Yes, musl and uclibc will need toolchain recompile for latest nightly. I changed the directory structure a bit.

rampageX, 2022-09-23:
nightly musl version is nice!

closed 2022-09-23 by rampageX